llvm/circt #4937: [ExportVerilog] Invalid part-select on a constant

xerpi, 2023-04-05:
The following MLIR:
```llvm
hw.module @test(..., %val0: i512, ...)
  %c0_i512 = hw.constant 0 : i512
  %c0_i32 = hw.constant 0 : i32
  %100 = comb.mux %13#6, %val0, %c0_i512 : i512
  %101 = sv.indexed_part_select %100[%c0_i32 : 64] : i512, i32
```
Generates the following SV:
```sv
wire [63:0]      _GEN_5 = _controller_out6 ? val0 : 512'h0[32'h0 +: 64];
```
Which Vitis `v++` doesn't like:
```
ERROR: [VPL 10-4982] syntax error near '[' [src/test_load_kernel_f1_s0_v0_v0.sv:316]
```
Probably the problem is at `512'h0[32'h0 +: 64];`.

uenoku, 2023-04-05:
The IR looks a bit weird to me. The first operand of`sv.indexed_part_select` must be an inout type but in this example it's a normal integer type. Could you share the entire MLIR file if you can share?

xerpi, 2023-04-05:
When emitting Verilog with `disallowMuxInlining = true`, the problem disappears because the generated Verilog looks like this:
```sv
wire [511:0]     _GEN_5 = _controller_out6 ? val0 : 512'h0;
wire [63:0]      _GEN_6 = _GEN_5[32'h0 +: 64];
```

But this makes the code much longer and difficult to read.
Also, this makes an error appear in the SV code lowered from FSMs:
```
ERROR: [VPL 10-1102] net data types must be 4 state [src/CalyxExtMemToAxi.sv:46]
```
The error is at the following line of `CalyxExtMemToAxi.sv`:
```sv
wire CalyxExtMemToAxi_state_t _GEN = calyx_read_en ? to_AR_HANDSHAKE : to_IDLE;
```

uenoku, 2023-04-05:
https://github.com/llvm/circt/commit/092d9b30bfc13ea04c2419dd7b14bc9a7bf8b473 should have fixed the issue. Could you try the latest commit? `ERROR: [VPL 10-1102] net data types must be 4 state [src/CalyxExtMemToAxi.sv:46]` looks different issue and probably it looks we have to create a temporary enum variable instead of a wire. Can you open a different issue?

closed 2023-04-05 by uenoku

xerpi, 2023-04-05:
Thank you very much!
Right now I'm trying the fix.
I'll also open an issue for the FSM problem

commit 092d9b30 (2023-04-05): [ExportVerilog] Fix `IndexedPartSelectOp` verilog emission

Close #4937.