llvm/circt #4948: [FIRRTL] Names destroyed during lowering

jackkoenig, 2023-04-06:
I'm mentioning this because it's an example on the Chisel website that is less-than-ideal 🙈 (See `Example3`): https://www.chisel-lang.org/chisel3/docs/explanations/naming.html

```
circuit Example3 :
  module Example3 :
    input clock : Clock
    input reset : UInt<1>
    input in : UInt<2>
    output out : UInt

    node _opt_T = mul(in, in)
    node _opt_T_1 = add(UInt<2>("h3"), _opt_T)
    node opt = tail(_opt_T_1, 1)
    node _out_T = add(opt, UInt<1>("h1"))
    node _out_T_1 = tail(_out_T, 1)
    out <= _out_T_1
```
firtool 1.37.0 gives
```verilog
// Generated by CIRCT firtool-1.37.0
module Example3(
  input        clock,
               reset,
  input  [1:0] in,
  output [3:0] out
);

  wire [3:0] _GEN = {2'h0, in};
  assign out = _GEN * _GEN + 4'h4;
endmodule
```

SFC gives:
```verilog
module Example3(
  input        clock,
  input        reset,
  input  [1:0] in,
  output [3:0] out
);
  wire [3:0] _opt_T = in * in;
  wire [3:0] opt = 4'h3 + _opt_T;
  assign out = opt + 4'h1;
endmodule
```

seldridge, 2023-04-06:
The output is possibly slightly better in debug mode as the named signals are then preserved (`firtool -O=debug`):

```verilog
module Example3(
  input        clock,
               reset,
  input  [1:0] in,
  output [3:0] out
);

  wire [3:0] _GEN = {2'h0, in};
  wire [3:0] opt = _GEN * _GEN + 4'h3;
  assign out = opt + 4'h1;
endmodule
```

However, this is hard because CIRCT really wants to optimize `a * a + 3 + 1` to `a * a + 4`. The `-O=release` output is effectively this optimization of: `out = in * in + 4`. The padding is annoying and maybe this could be code gen'd better.

Given the constraints of Verilog around padding it may be possible to move instead towards:

```
wire [3:0] _opt_T = in * in;
assign out = _opt_T + 4'h4;
```

However, I think `opt` is going to get clobbered in release mode.

mikeurbach, 2023-04-06:
One thing we do have up our sleeve is our wireSpillingHeuristic, which can emit wires for names when present, even in release mode. This is not the default right now.

If we turn this on, _and_ add a change I'm experimenting with to disable inlining sub-expressions when sub-expressions have a good name, we can get the current debug mode output shown above.

The temporary for the concat is unfortunate, and I'm separately looking at if there is a better way to handle this.

jackkoenig, 2023-04-06:
```
wire [3:0] _opt_T = in * in;
assign out = _opt_T + 4'h4;
```
I think the above is solid, the reasons make sense. I only included the specific example because it's one we have on the Chisel website and thus it was a bit... striking. I've separately improved the example to better show what we were trying to show anyway.