llvm/circt #5008: [FIRRTL][IMCP] Don't const-prop through read of dontTouch'd output port

dtzSiFive, 2023-04-11:
Example input:

```firrtl
circuit Producer : %[[
 { "class": "firrtl.transforms.DontTouchAnnotation",
   "target": "~Producer|Producer>foo" },
 { "class": "firrtl.transforms.DontTouchAnnotation",
   "target": "~Producer|Producer>out" }
]]
  module Producer:
    output out: UInt<5>
    
    ; Simple constant output.
    node n = UInt<5>(1)
    out <= n

    ; Output port drives internal wire.
    wire foo : UInt<5>
    foo <= out
```

Current output (as well as back to, randomly, 1.25.0):

```systemverilog
// Generated by CIRCT 1.37.0g20230411_c9c3591
module Producer(
  output [4:0] out
);

  wire [4:0] foo = 5'h1;
  assign out = 5'h1;
endmodule
```

uenoku, 2023-04-12:
Actually IMCP is correct. This is entirely an issue of LowerToHW which drops symbols when creating temporaries for output wires. 
```mlir
// -----// IR Dump After IMConstProp (firrtl-imconstprop) //----- //
firrtl.circuit "Producer" {
  firrtl.module @Producer(out %out: !firrtl.uint<5> [{class = "firrtl.transforms.DontTouchAnnotation"}]) attributes {convention = #firrtl<convention scalarized>} {
    %c1_ui5 = firrtl.constant 1 : !firrtl.uint<5>
    firrtl.strictconnect %out, %c1_ui5 : !firrtl.uint<5>
    %foo = firrtl.wire {annotations = [{class = "firrtl.transforms.DontTouchAnnotation"}]} : !firrtl.uint<5>
    firrtl.strictconnect %foo, %out : !firrtl.uint<5>
  }
}
```

dtzSiFive, 2023-04-12:
Okay, great, thanks for taking a look.  That's #5011, closing this.

closed 2023-04-12 by dtzSiFive