llvm/circt #5014: [PyCDE] No module named "numpy"/"jinja2" during compilation

VMois, 2023-04-12:
When I try to compile PyCDE (cannot install cause using Apple M1), I have the following error (in many places):

```
Traceback (most recent call last):
        File "/Users/vmois/Projects/circt/frontends/PyCDE/test/test_xrt.py", line 20, in <module>
          import pycde
        File "/Users/vmois/Projects/circt/frontends/PyCDE/build/cmake_build/tools/circt/python_packages/pycde/pycde/__init__.py", line 22, in <module>
          from .module import (generator, modparams, Module)
        File "/Users/vmois/Projects/circt/frontends/PyCDE/build/cmake_build/tools/circt/python_packages/pycde/pycde/module.py", line 11, in <module>
          from .signals import ClockSignal, Signal, _FromCirctValue
        File "/Users/vmois/Projects/circt/frontends/PyCDE/build/cmake_build/tools/circt/python_packages/pycde/pycde/signals.py", line 18, in <module>
          import numpy as np
      ModuleNotFoundError: No module named 'numpy'
```
After checking required packages, I can see that numpy is present:

https://github.com/llvm/circt/blob/c9c359159ad977892e13865e83c8cd0951b21e74/frontends/PyCDE/pyproject.toml#L2-L17

I have installed numpy manually (`pip install numpy`), but now a different package is missing - `jinja2`. After manually installing `jinja2`, I managed to build pycde and it works! But, strangely, packages are not found while specified in `required`. Any ideas why it can be?

I am using conda env and Python 3.8.

teqdruid, 2023-04-12:
Probably 'cause someone (me) who's not at all familiar with the relevant technologies last touched it. @mikeurbach Do you have any idea why the deps are not being installed during the build?

It could be that they're pre-installed on the docker image for linux which cibuildwheel uses so we don't run into this problem...

VMois, 2023-04-12:
I forgot to mention that the current `pip` version (23.0) doesn't support `--use-feature=in-tree-build`, so I have removed this flag and just run `pip install`. According to this [SO answer](https://stackoverflow.com/a/73908418/7450581), `in-tree-build` is the default behavior now. 

I will try to compile it again with an older pip version and `in-tree-build` flag and see what happens.

mikeurbach, 2023-04-12:
I don't expect `--use-feature=in-tree-build` to be an issue, since I believe it should be essentially the same as using the latest `pip` in this regard.

For one thing, that section of the pyproject.toml is about declaring build time dependencies: https://pip.pypa.io/en/stable/reference/build-system/pyproject-toml/#build-time-dependencies. But I think the failure happens at run time. Or more precisely, when running the pycde tests while building the wheel. I guess adding these as build time dependencies doesn't really work for that case.

I see these are also added as `install_requires` in setup.py: https://github.com/llvm/circt/blob/c9c359159ad977892e13865e83c8cd0951b21e74/frontends/PyCDE/setup.py#L112

I would have thought that should make sure they are available when building the wheel. I'll need to try this out on a clean environment to see where it fails.

I'll try again on Python 3.8 and pip 23 to see if I can reproduce, and if there is a way to better express the dependencies I will add it.

teqdruid, 2023-04-12:
Thanks for digging into this @mikeurbach

mikeurbach, 2023-04-12:
Yeah I can reproduce this. It's weird that the having the requirements in pyproject.toml under build-system.requires doesn't work, but maybe there are some shenanigans related to the cmake build and running the tests under cmake. I'm trying to add them under `setup_requires` as well as `install_requires` in setup.py, and I think this should work.

mikeurbach, 2023-04-12:
Nope... that wasn't it either. In fact, on the main branch, I can see the dependencies are installed:

```
$ pip install frontends/PyCDE
Processing ./frontends/PyCDE
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting numpy
  Using cached numpy-1.24.2-cp38-cp38-macosx_11_0_arm64.whl (13.8 MB)
Collecting jinja2
  Using cached Jinja2-3.1.2-py3-none-any.whl (133 kB)
Collecting MarkupSafe>=2.0
  Using cached MarkupSafe-2.1.2-cp38-cp38-macosx_10_9_universal2.whl (17 kB)
Building wheels for collected packages: pycde
  Building wheel for pycde (pyproject.toml) ... error
```

I also tried adding these dependencies using the `project.dependencies` key in pyproject.toml, but that also fails.

teqdruid, 2023-04-17:
> I will try to compile it again with an older pip version and `in-tree-build` flag and see what happens.

You can just omit this flag for newer versions.