llvm/circt #5043: [CI] Windows build and test fails randomly

maerhart, 2023-04-14:
I've had two seemingly random Windows CI failures on main recently (e7f306d and 15b43af). Both failing some firtool test (but a different one). Anyone a clue why that is? Is the build unstable or some UB somewhere?

teqdruid, 2023-04-17:
We've had problems like this in the past. I suspect it's resource usage (i.e. out of memory or disk space).

seldridge, 2023-11-11:
Closing this as the most recent maybe fix discussed in https://github.com/llvm/circt/issues/6193 is to bump the Windows runners. We can open another issue or revive this one in the future. However, I just want to keep the Windows build failures linked together and not have a bunch of ones open. We can always find them via search.

closed 2023-11-11 by seldridge