llvm/circt #5052: [PyCDE] 'value' must be either ir.Value or Signal when using many data inputs in Mux construct

VMois, 2023-04-14:
Hi. After installing the PyCDE, I spent some time understanding the basics. I decided to build a simple calculator that has 4 operations hence need to use Mux with more than 2 values. I found already implemented Mux here:

https://github.com/llvm/circt/blob/6ff1ddc0ea77d8adc47df1f0efb809fad1bace3d/frontends/PyCDE/src/constructs.py#L181

This simple mux with two data inputs runs and procudes correct SV.

```python
from pycde import Input, Output, Module, System
from pycde import generator
from pycde.types import Bits
from pycde.constructs import Mux

class SimpleMux(Module):
    op = Input(Bits(1))
    a = Input(Bits(32))
    b = Input(Bits(32))
    out = Output(Bits(32))

    @generator
    def construct(self):
        self.out = Mux(self.op, self.a, self.b)

system = System([SimpleMux], name="ExampleSystem", output_directory="exsys")
system.compile()
```

After checking how `Mux` works, I decided to add two more operations and see if it works.

```python
class SimpleMux(Module):
    op = Input(Bits(2))
    a = Input(Bits(32))
    b = Input(Bits(32))
    c = Input(Bits(32))
    d = Input(Bits(32))
    out = Output(Bits(32))

    @generator
    def construct(self):
        self.out = Mux(self.op, self.a, self.b, self.c, self.d)
```

The first (solvable) issue was missing `type` argument here:

https://github.com/llvm/circt/blob/6ff1ddc0ea77d8adc47df1f0efb809fad1bace3d/frontends/PyCDE/src/constructs.py#L194

I managed to fix it by directly modifying PyCDE code (as a tmp workaround):

```python
ArraySignal(..., type=data_inputs[0].type)
```

When I ran it again this error occurred:

```
File "/Users/vmois/miniforge3/envs/circt/lib/python3.8/site-packages/pycde/signals.py", line 42, in __init__
    assert False, "'value' must be either ir.Value or Signal"
AssertionError: 'value' must be either ir.Value or Signal
```

`ArraySignal` inherets from `Signal` so when object is initializing `Signal` init method is used.

https://github.com/llvm/circt/blob/6ff1ddc0ea77d8adc47df1f0efb809fad1bace3d/frontends/PyCDE/src/signals.py#L32-L43

After checking with a debugger, I could see that value passed to `Signal.init` is a list of `BitsSignal` objects. And it can only work with a single values.

```
-> elif isinstance(value, Signal):
(Pdb) value
(<pycde.signals.BitsSignal object at 0x10586cd90>, <pycde.signals.BitsSignal object at 0x10586ce50>, <pycde.signals.BitsSignal object at 0x10586cf10>, <pycde.signals.BitsSignal object at 0x10586cee0>)
(Pdb) isinstance(value, Signal)
False
```

Am I doing something wrong with PyCDE API, or is it really a bug? Thank you very much.

VMois, 2023-04-14:
I managed to patch it. But not sure if my approach is breaking things cause I don't know how to use PyCDE or fix them. I decided to use `Signal.create()` in `Mux`:

```python
a = ArraySignal(Signal.create([*data_inputs]), type=data_inputs[0].type)
```

After rerunning 4 input MUX compilation, I got a different error:

```
File "/Users/vmois/miniforge3/envs/circt/lib/python3.8/site-packages/pycde/signals.py", line 505, in __getitem__
    _validate_idx(self.type.width, idx)
AttributeError: 'Bits' object has no attribute 'size'
```

I switched `size` to `width`. This is also what the `Mux` code is using:

https://github.com/llvm/circt/blob/6ff1ddc0ea77d8adc47df1f0efb809fad1bace3d/frontends/PyCDE/src/constructs.py#L188

After that, things worked, and I got this output:

```
// Generated by CIRCT unknown git version
module SimpleMux
  #(parameter __INST_HIER = "INSTANTIATE_WITH_INSTANCE_PATH") (	// simple_mux.py:6
  input  [1:0]  op,
  input  [31:0] a,
                b,
                c,
                d,
  output [31:0] out
);

  wire [3:0][31:0] arr_a_b_c_d = {{d}, {c}, {b}, {a}};	// simple_mux.py:16
  assign out = arr_a_b_c_d[op];	// simple_mux.py:15, :16
endmodule
```

The SystemVerilog looks correct to me.

My question is again whether I am missing something or if it is an actual bug. Thanks!

teqdruid, 2023-04-17:
Yep -- this was a bug. Fixed by https://github.com/llvm/circt/commit/81b5e40608619e487e1cefb63b027500aac24f27. Thanks for reporting it! I revamped the signal system awhile back and didn't have a test for muxes with >2 inputs. (D'oh!)

In general, the `Signal.create` method is used to create signals with the correct type. What you want to do with the ArraySignal constructor isn't unreasonable though.

closed 2023-04-17 by teqdruid