llvm/circt #5056: Question about release assets

leonardt, 2023-04-17:
I was looking through https://github.com/llvm/circt/releases and I saw that different versions of firtool releases provide different assets.  For example, `1.36` provides `circt-bin`, whereas `1.37` provides `firrtl-bin`.  Is there a plan to standardized release assets?  In particular, I'm interested in a regular `circt-bin` release to integrate into downstream CI and to simplify installation for downstream users.

mikeurbach, 2023-04-21:
The releases are currently just packaging firtool based on the firtool-* tags, and those are used for Chisel among other things. This will be the plan for those releases moving forward, and we have removed the other assets from the release tarball besides the firtool binary. There will be regular firtool releases, but unless you are interested in firtool, those might not be useful to you.

What binaries for CIRCT do you want to integrate in downstream CI and provide for users? It is usually best to wrap up the compiler pipeline you care about into a tool (like firtool, or hlstool, etc.), if it is something for end users. circt-opt and friends are good for compiler developers, but not great for end users.

leonardt, 2023-04-21:
We use `circt-opt` to convert [magma](https://github.com/phanrahan/magma) generated MLIR to verilog, packaging a pre-compiled binary does seem like a viable option (we could distribute it as a pip package wheel), but it seems likely useful to distribute this independent of magma since we don't really have a custom version and I'd imagine their might be other tools that could share a dependency on it.

We have done some work on distributing python binary wheels for multiple platforms so we could take a look into setting up a CI pipeline for that.

mikeurbach, 2023-04-21:
Another option might be for Magma to use CIRCT as a Python library.

I remember when we first talked about Magma and CIRCT, the Python bindings were very new, and it was advised to just print out CIRCT IRs textually. Well, time has passed, and the CIRCT Python bindings are much more stable, so perhaps it is interesting to look into using them from Magma.

Regardless of how the IR is produced, it should be possible to adopt CIRCT as a Python dependency in Magma, just to call the relevant passes you use circt-opt for today. Basically, you'd use the existing serializers to print CIRCT IR, parse it back into an in-memory object, and use the Python APIs to run passes. We ship CIRCT to PyPI when the current releases are built: https://pypi.org/project/circt/.

This could allow you to have a simple installation of the CIRCT tools, and if you eventually use the Python bindings to build IR, you can avoid the need to serialize to a string as well.