llvm/circt #5066: [FIRRTL] Dut and Harness separation flag for `firtool`

joey0320, 2023-04-21:
Hi All,

We were wondering if it would be possible to support a flag in `firtool` that uniquifies the modules under the `Dut` and `Harness`. When running timing annotated simulations, after the `Dut` is synthesized the modules that are common to the `Dut` and `Harness` will collide and cause simulation to fail.

The related issue is ucb-bar/chipyard#1388 and ucb-bar/chipyard#1422 is a temporary workaround which post-processes the files using a python script.

Thanks!

Related people to this issue
@harrisonliew
@abejgonzalez
@allpan3

davidmlw, 2023-07-03:
Is there some specific requirements for this? I think Chisel and firtool view Model and DUT the same, both as a synthesizable module. So how firtool knows the boundary? Need it add new annotation in FIRRTL IR?

tymcauley, 2023-11-05:
@joey0320 I think the PR link in the issue description should be https://github.com/ucb-bar/chipyard/pull/1442 rather than 1422.