llvm/circt #5122: [FIRRTL] Add Probes to Exporter

seldridge, 2023-05-03:
The FIRRTL exporter does not currently implement `probe` and associated reference type operations. Add support for these and make sure they round-trip correctly.

Consider the following:

```
circuit Foo:
  module Foo:
    output a: Probe<UInt<1>>
    output b: RWProbe<UInt<1>>

    wire x: UInt<1>

    define a = probe(x)
    define b = rwprobe(x)
```

This currently hits an unreachable condition if you try to export it:

```
# firtool Bar.fir -parse-only | circt-translate -export-firrtl
circuit Foo :
  module Foo :
    output a : all types should be implemented
UNREACHABLE executed at /Users/schuylere/repos/github.com/llvm/circt/lib/Dialect/FIRRTL/Export/FIREmitter.cpp:787!
```

seldridge, 2023-05-08:
I fixed this except for `rwprobe` in https://github.com/llvm/circt/commit/50636dde95e2c959e737a2328099e7e9e1dc9cec.

dtzSiFive, 2023-05-10:
Thanks for pushing on this!! :bowing_man: 

Few other ops for complete FIRRTL 2.0 probe/rwprobe import/export support:

* {force,release}{,_initial}
* ref.sub
* internalPaths? (ref statements)

seldridge, 2023-05-10:
Added `rwprobe` in https://github.com/llvm/circt/commit/cd7f0d17e241c335fbd17d76804a1b94eff716b8 and https://github.com/llvm/circt/commit/e0d380299b60f489fd8216a332b7d3319995db32.

dtzSiFive, 2023-05-16:
ref.sub handled in #5179 .

dtzSiFive, 2023-08-28:
Force/release: 65546f4c61359041a3d1b07fe1851ed6492700b7
e0d380299b60f489fd8216a332b7d3319995db32
RefCast, RWProbeOp (RWProbe to local inner symbol), and improved Forceable support in: #5779.

I think everything is handled other than possibly ref statements/`internalPaths` (on external modules).

cc https://github.com/llvm/circt/issues/5205 for probe+emitter-related bug.

seldridge, 2024-02-10:
Calling this done. 😉

closed 2024-02-10 by seldridge