llvm/circt #5154: [lower-scf-to-fsm] error: Operation 'calyx.par' not supported for FSM compilation

zzy666666zzy, 2023-05-08:
[gemm8x8_no_unroll_scf_for.mlir.txt](https://github.com/llvm/circt/files/11425655/gemm8x8_no_unroll_scf_for.mlir.txt)

Hi there,

Thanks for contributing to this open-source agile hardware development tool.

I tried to lower an MLIR of a general 8 times 8 matrix multiplication. I got an error '**Operation 'calyx.par' not supported for FSM compilation**' when lowering my calyx dialects to fsm. I wonder is there any solution to address this issue? 

Attached please find the MLIR file.

Many thanks!

zzy666666zzy, 2023-05-08:
This is the file I used for lowering scf.while to calyx. 
Thanks.
[gemm8x8_no_unroll_scf_while.mlir.txt](https://github.com/llvm/circt/files/11425708/gemm8x8_no_unroll_scf_while.mlir.txt)

rachitnigam, 2023-05-09:
Yup, compilation is not supported by the MLIR flow for Calyx. One fix is using the [Calyx native compiler][cn]. You can generate native compatible IL [using `circt-translate`](https://docs.calyxir.org/fud/circt.html#mlir-to-native-representation):
```
circt/bin/circt-translate --export-calyx <file>
```
And then use the native compiler to generate lowered Calyx code (from the calyx compiler repo):
```
cargo run -- <calyx-native-file> -b mlir > compiled.mlir
```
You can compile the resulting code directly to Verilog if you'd like instead:
```
cargo run -- <calyx-native-file> 
```

Let me know if you run into any more problems and feel free to ask more questions by [opening issues in the Calyx repo][cn].

[cn]: https://github.com/cucapra/calyx

rachitnigam, 2023-05-09:
Here is what I ran on the program with `scf.while` (`scf.for` is marked illegal by `scf-to-calyx`):
```
./build/bin/circt-opt --lower-scf-to-calyx gemm-while.mlir | ./build/bin/circt-translate --export-calyx 
```
This generates output that can be used by the native compiler for processing. Then, you can install the [Calyx native compiler using its cargo crate][calyx-crate] which installs the `futil` binary:
```
cargo install calyx
```
Then, we can generate Verilog by running:
```
 ./build/bin/circt-opt --lower-scf-to-calyx gemm-while.mlir | ./build/bin/circt-translate --export-calyx | futil -l ../calyx -b verilog
```

[calyx-crate]: https://crates.io/crates/calyx

mortbopet, 2023-05-09:
Closing this since it's a duplicate of https://github.com/llvm/circt/issues/4606. Help is very much appreciated!

closed 2023-05-09 by mortbopet