llvm/circt #5171: [ExportVerilog] Option to disable always comb fusion

leonardt, 2023-05-10:
MLIR
```mlir
module attributes {circt.loweringOptions = "locationInfoStyle=none"} {
    hw.module @test_when_alwcomb_order_2(%I: i8, %S: i2, %CLK: i1) -> (O: i8) {
        %0 = comb.extract %S from 0 : (i2) -> i1
        %1 = comb.extract %S from 1 : (i2) -> i1
        %2 = hw.constant 0 : i8
        %5 = sv.reg : !hw.inout<i8>
        %4 = sv.read_inout %5 : !hw.inout<i8>
        sv.alwayscomb {
            sv.if %1 {
                sv.bpassign %5, %2 : i8
            } else {
                sv.bpassign %5, %3 : i8
            }
        }
        %7 = sv.reg name "a" : !hw.inout<i8>
        sv.alwaysff(posedge %CLK) {
            sv.passign %7, %4 : i8
        }
        %8 = hw.constant 0 : i8
        sv.initial {
            sv.bpassign %7, %8 : i8
        }
        %6 = sv.read_inout %7 : !hw.inout<i8>
        %10 = sv.reg : !hw.inout<i8>
        %3 = sv.read_inout %10 : !hw.inout<i8>
        %11 = sv.reg : !hw.inout<i8>
        %9 = sv.read_inout %11 : !hw.inout<i8>
        sv.alwayscomb {
            sv.bpassign %10, %6 : i8
            sv.bpassign %11, %6 : i8
            sv.if %0 {
                sv.bpassign %10, %I : i8
                sv.bpassign %11, %I : i8
            }
        }
        %13 = sv.reg name "b" : !hw.inout<i8>
        sv.alwaysff(posedge %CLK) {
            sv.passign %13, %9 : i8
        }
        sv.initial {
            sv.bpassign %13, %8 : i8
        }
        %12 = sv.read_inout %13 : !hw.inout<i8>
        hw.output %12 : i8
    }
}
```

produces

```verilog
// Generated by CIRCT circtorg-0.0.0-1773-g7abbc4313
module test_when_alwcomb_order_2(
  input  [7:0] I,
  input  [1:0] S,
  input        CLK,
  output [7:0] O
);

  reg [7:0] _GEN;
  reg [7:0] a;
  reg [7:0] _GEN_0;
  reg [7:0] _GEN_1;
  always_comb begin
    if (S[1])
      _GEN = 8'h0;
    else
      _GEN = _GEN_0;
    _GEN_0 = a;
    _GEN_1 = a;
    if (S[0]) begin
      _GEN_0 = I;
      _GEN_1 = I;
    end
  end // always_comb
  reg [7:0] b;
  always_ff @(posedge CLK) begin
    a <= _GEN;
    b <= _GEN_1;
  end // always_ff @(posedge)
  initial begin
    a = 8'h0;
    b = 8'h0;
  end // initial
  assign O = b;
endmodule
```

Notice that `_GEN_0` is referenced before it is assigned.  In the original MLIR code, the reference and assignment are in separate `sv.alwayscomb` blocks, which would imply a dependency between the two blocks, but in the generated verilog the always comb blocks are fused which leads to a read before write within a block.  It seems like a simple option to avoid this issue would be to add a flag to disable the fusion of `sv.alwayscomb` blocks in the verilog code generation.

See https://github.com/phanrahan/magma/issues/1269 for more context.

mikeurbach, 2023-05-10:
A flag to disable this behavior seems reasonable; we have all sorts of flags like this for controlling these kinds of things. It also seems like a bug that should be fixed... If this were fixed, would you still want the flag, or is the fusion (when legal) desirable in your use case? Just trying to plan a bit, since it seems like we should fix the bug regardless.

uenoku, 2023-05-11:
Thanks, then in genreal it's illegal to fuse always_comb? Actually HWCleanUp pass is performing that so it's reasonable to just remove them. There has been no user of `always_comb` so I think it's ok to remove them.

leonardt, 2023-05-11:
I think legal fusion would be fine and likely desirable in most cases

leonardt, 2023-05-11:
Another, perhaps useful option, would be to allow the user to annotate certain blocks to prevent/allow fusion

uenoku, 2023-05-17:
> I think legal fusion would be fine and likely desirable in most cases

Yes, I agree but I guess it would require fair amount of work to perform safely so I feel it's

> Another, perhaps useful option, would be to allow the user to annotate certain blocks to prevent/allow fusion

Probably useful, but I feel if the frontend knew it's safe to fuse always comb statements, then I wonder it's better for frontend to emit such IR in the first place as the dependency analysis is much easier to perform at more high level abstraction.

commit e780e3e7 (2023-05-23): [HWCleanUp] Remove always_comb (#5175)

It's illegal to merge different always_comb in general as it might implicitly 
have data dependency each other. Fix #5171
closed 2023-05-23 by uenoku