llvm/circt #5173: [CI] Add more wheel configurations

rsetaluri, 2023-05-10:
I see that only a few configurations are used for building python wheels:

https://github.com/llvm/circt/blob/de03ab500b1803eec73c4418af269151342094d6/.github/workflows/uploadWheels.yml#L14-L19

We would like to add python3.8 as well as OSX builds to this. I believe python3.8 would be straightforward, but OSX might require more work (from cc @leonardt 's experience).

mikeurbach, 2023-05-10:
I think it should be possible to use OSX, but I haven't looked into it yet, and I haven't been testing manylinux with OSX in general. It should be supported on both Intel and Apple silicon according to https://cibuildwheel.readthedocs.io/en/stable/ (apparently with cross compiling for Apple silicon).

rsetaluri, 2023-05-16:
@mikeurbach I also see that the wheel is not always included in the release assets? What condition controls that? Would it be possible for it to be released more frequently (or always)? (For example in [Firtool Release 1.42.0](https://github.com/llvm/circt/releases/tag/firtool-1.42.0), I see that there are wheels, even though my CI change is included (#5174)).

mikeurbach, 2023-05-16:
That would be these conditions:

https://github.com/llvm/circt/blob/a3ac707ef5ab29adf1d0ff646e2248a4f73b6f39/.github/workflows/uploadWheels.yml#L48-L57

Basically, if not running on a tag (testing the workflow, manual triggers, etc.), it will upload the wheels into the release assets. But for a tagged release, a wheel is uploaded directly to PyPI: https://pypi.org/project/circt/#history (note that 1.41.0 was DOA because of an upstream MLIR issue that prevented the wheels to build).

As of now, the Python wheels are tied to the Firtool releases, because these are made weekly with extra testing beyond CIRCT CI. They don't have to be, but that is the current setup.

rsetaluri, 2023-05-17:
I see -- the differentiation of pypi vs. artifact had me confused; thanks for clarifying. I think weekly releases to pypi are plenty.

@mikeurbach: I tried the latest pypi release (1.42.0) on centos8 with python3.8. Here is the output:

```
(env) $ python -m pip install circt
Collecting circt
  Using cached circt-1.42.0-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (8.9 MB)
Installing collected packages: circt
Successfully installed circt-1.42.0

(env) $ python -c 'import circt'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/rajsekhar/env/lib/python3.8/site-packages/circt/__init__.py", line 5, in <module>
    from ._mlir_libs._circt import *
  File "/home/rajsekhar/env/lib/python3.8/site-packages/circt/_mlir_libs/__init__.py", line 128, in <module>
    _site_initialize()
  File "/home/rajsekhar/env/lib/python3.8/site-packages/circt/_mlir_libs/__init__.py", line 56, in _site_initialize
    from ._mlir import ir
ImportError: cannot import name 'ir' from 'circt._mlir_libs._mlir' (unknown location)
```

Here is the system information

```
(env) $ python --version
Python 3.8.6

(env) $ uname -a
Linux snc-arvrusr9178.thefacebook.com 4.18.0-408.el8.x86_64 #1 SMP Mon Jul 18 17:42:52 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

(env) $ cat /etc/os-release 
NAME="CentOS Stream"
VERSION="8"
ID="centos"
ID_LIKE="rhel fedora"
VERSION_ID="8"
PLATFORM_ID="platform:el8"
PRETTY_NAME="CentOS Stream 8"
ANSI_COLOR="0;31"
CPE_NAME="cpe:/o:centos:centos:8"
HOME_URL="https://centos.org/"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_SUPPORT_PRODUCT="Red Hat Enterprise Linux 8"
REDHAT_SUPPORT_PRODUCT_VERSION="CentOS Stream"
```

rsetaluri, 2023-05-17:
I see -- the differentiation of pypi vs. artifact had me confused; thanks for clarifying. I think weekly releases to pypi are plenty.

@mikeurbach: I tried the latest pypi release (1.42.0) on centos8 with python3.8. Here is the output:

```
(env) $ python -m pip install circt
Collecting circt
  Using cached circt-1.42.0-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (8.9 MB)
Installing collected packages: circt
Successfully installed circt-1.42.0

(env) $ python -c 'import circt'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/env/lib/python3.8/site-packages/circt/__init__.py", line 5, in <module>
    from ._mlir_libs._circt import *
  File "/home/env/lib/python3.8/site-packages/circt/_mlir_libs/__init__.py", line 128, in <module>
    _site_initialize()
  File "/home/env/lib/python3.8/site-packages/circt/_mlir_libs/__init__.py", line 56, in _site_initialize
    from ._mlir import ir
ImportError: cannot import name 'ir' from 'circt._mlir_libs._mlir' (unknown location)
```

Here is the system information

```
(env) $ python --version
Python 3.8.6

(env) $ cat /etc/os-release 
NAME="CentOS Stream"
VERSION="8"
ID="centos"
ID_LIKE="rhel fedora"
VERSION_ID="8"
PLATFORM_ID="platform:el8"
PRETTY_NAME="CentOS Stream 8"
ANSI_COLOR="0;31"
CPE_NAME="cpe:/o:centos:centos:8"
HOME_URL="https://centos.org/"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_SUPPORT_PRODUCT="Red Hat Enterprise Linux 8"
REDHAT_SUPPORT_PRODUCT_VERSION="CentOS Stream"
```

mikeurbach, 2023-05-17:
Hmm, let me check that out, I have seen that error in local development when I got something wrong, but not from a wheel.

rsetaluri, 2023-05-17:
@mikeurbach thanks for looking into it. Turns out there was something wrong with the python version/setup. Using Python 3.8.13 it was able to work! Next step macos...

mikeurbach, 2023-05-17:
Awesome, glad it was a passing thing. Re: MacOS, I know it can work because I have built the Python bindings locally on a Mac, but I haven't tried cibuildwheel's support for MacOS.

leonardt, 2023-05-17:
I believe the main work would be setting up the non-python related portion of the CIRCT build in the GitHub actions macOS environment, which likely boils down to just making sure we get the appropriate dependencies installed.  Unless cibuildwheel can build a macOS wheel from a linux environment?  Otherwise, we've used cibuildwheel in a macOS environment to build the universal (intel/arm) wheels without a problem (perhaps https://github.com/Kuree/uinspect/blob/master/.github/workflows/ci.yml as an example may be helpful if we run into issues)

rsetaluri, 2023-07-26:
We now have (python3.7, python3.8, python3.10) x (manylinux, macos) working

closed 2023-07-26 by rsetaluri