llvm/circt #5182: [HW][MSFT] Replace GlobalRefOp with HierPathOp

seldridge, 2023-05-12:
Now that `HierPathOp` has fully migrated into HW Dialect from FIRRTL, it should be a complete replacement for `GlobalRefOp`. The only actual load-bearing uses appear to be in the MSFT Dialect.

teqdruid, 2023-05-16:
Is there some documentation for `HierPathOp` a la the [`GlobalRefOp`](https://circt.llvm.org/docs/Dialects/HW/RationaleHW/#globalrefop)?

Also, I'm guessing that hierarchy-modifying passes must just be cognizant of `HierPathOp` and modify them accordingly? I see there's an `NLATable` helper class in the FIRRTL dialect which seems to be only loosely connected to FIRRTL. Could this be moved into the HW dialect?

seldridge, 2023-05-18:
Yeah, no real documentation. Roughly, it's the same as `GlobalRefOp` except that it has no breadcrumbs (`circt.globalref` attributes) scattered throughout the circuit.

Yes, passes must be cognizant of the hierarchical paths and update them accordingly.

Yes, `NLATable` is intended to help with this.

Yes, that should be moved to HW Dialect. This was always the intent. I can't recall why, but I think there were some hurdles with this. (I think @dtzSiFive may recall?) Alternatively, I may have just never done it. 😅 

The motivation for dropping the breadcrumbs was that any pass that wanted to use the breadcrumbs had to go walk the whole circuit. Hence, the breadcrumbs were basically pointless. This was rolled into `NLATable` which does the walk, finds everything for you, helps you update things, and doesn't pollute the circuit with tons of attributes.

teqdruid, 2023-09-21:
Closed by #6168 . Referenced the wrong issue in the commit.

closed 2023-09-21 by teqdruid