llvm/circt #5184: How to generate generate the generic form of operations for the dialects in CIRCT?

Stevengre, 2023-05-12:
Hello everyone! I'm new to MLIR and CIRCT, and I'm trying to generate the generic form of operations for the HW dialect using `mlir-opt --mlir-print-op-generic`. However, it seems that the HW dialect is not registered, as I encountered the following error:

```bash
mlir-opt --mlir-print-op-generic circt/test/Dialect/HW/basic.mlir

circt/test/Dialect/HW/basic.mlir:4:1: error: Dialect `hw' not found for custom op 'hw.module'
hw.module @test1(%arg0: i3, %arg1: i1, %arg2: !hw.array<1000xi8>) -> (result: i50) {
^
circt/test/Dialect/HW/basic.mlir:4:1: note: Registered dialects: acc, affine, amdgpu, amx, arith, arm_neon, arm_sve, async, bufferization, builtin, cf, complex, dlti, emitc, func, gpu, index, irdl, linalg, llvm, math, memref, ml_program, nvgpu, nvvm, omp, pdl, pdl_interp, quant, rocdl, scf, shape, sparse_tensor, spirv, tensor, test, test_dyn, tosa, transform, vector, x86vector; for more info on dialect registration see https://mlir.llvm.org/getting_started/Faq/#registered-loaded-dependent-whats-up-with-dialects-management
```

Could you please guide me on how to generate the generic form of operations for the dialects in CIRCT? Thank you in advance!

Stevengre, 2023-05-12:
Thank you so much! It does work for me! So is circle-opt providing all MLIR options?

uenoku, 2023-05-12:
Yes, circt-opt does. Also firtool should also accetpt most of mlir-opt options.