llvm/circt #5191: [CI] Windows build failure?

dtzSiFive, 2023-05-12:
https://github.com/llvm/circt/actions/runs/4959723038/jobs/8874229998

```
FAIL: CIRCT :: firtool/convention.fir (490 of 519)
******************** TEST 'CIRCT :: firtool/convention.fir' FAILED ********************
Script:
--
: 'RUN: at line 1';   d:\a\circt\circt\build_release\bin\firtool.exe D:\a\circt\circt\test\firtool\convention.fir --format=fir --parse-only --scalarize-top-module=false --scalarize-ext-modules=false --preserve-aggregate=all | d:\a\circt\circt\llvm\install\bin\filecheck.exe --check-prefix=SCALARIZE_NONE D:\a\circt\circt\test\firtool\convention.fir
: 'RUN: at line 2';   d:\a\circt\circt\build_release\bin\firtool.exe D:\a\circt\circt\test\firtool\convention.fir --format=fir --parse-only --scalarize-top-module=true  --scalarize-ext-modules=false --preserve-aggregate=all | d:\a\circt\circt\llvm\install\bin\filecheck.exe --check-prefix=SCALARIZE_TOP  D:\a\circt\circt\test\firtool\convention.fir
: 'RUN: at line 3';   d:\a\circt\circt\build_release\bin\firtool.exe D:\a\circt\circt\test\firtool\convention.fir --format=fir --parse-only --scalarize-top-module=false --scalarize-ext-modules=true  --preserve-aggregate=all | d:\a\circt\circt\llvm\install\bin\filecheck.exe --check-prefix=SCALARIZE_EXT  D:\a\circt\circt\test\firtool\convention.fir
: 'RUN: at line 4';   d:\a\circt\circt\build_release\bin\firtool.exe D:\a\circt\circt\test\firtool\convention.fir --format=fir --parse-only --scalarize-top-module=true  --scalarize-ext-modules=true  --preserve-aggregate=all | d:\a\circt\circt\llvm\install\bin\filecheck.exe --check-prefix=SCALARIZE_BOTH D:\a\circt\circt\test\firtool\convention.fir
--
Exit Code: 3221225477

Command Output (stdout):
--
$ ":" "RUN: at line 1"
$ "d:\a\circt\circt\build_release\bin\firtool.exe" "D:\a\circt\circt\test\firtool\convention.fir" "--format=fir" "--parse-only" "--scalarize-top-module=false" "--scalarize-ext-modules=false" "--preserve-aggregate=all"
$ "d:\a\circt\circt\llvm\install\bin\filecheck.exe" "--check-prefix=SCALARIZE_NONE" "D:\a\circt\circt\test\firtool\convention.fir"
$ ":" "RUN: at line 2"
$ "d:\a\circt\circt\build_release\bin\firtool.exe" "D:\a\circt\circt\test\firtool\convention.fir" "--format=fir" "--parse-only" "--scalarize-top-module=true" "--scalarize-ext-modules=false" "--preserve-aggregate=all"
$ "d:\a\circt\circt\llvm\install\bin\filecheck.exe" "--check-prefix=SCALARIZE_TOP" "D:\a\circt\circt\test\firtool\convention.fir"
$ ":" "RUN: at line 3"
$ "d:\a\circt\circt\build_release\bin\firtool.exe" "D:\a\circt\circt\test\firtool\convention.fir" "--format=fir" "--parse-only" "--scalarize-top-module=false" "--scalarize-ext-modules=true" "--preserve-aggregate=all"
$ "d:\a\circt\circt\llvm\install\bin\filecheck.exe" "--check-prefix=SCALARIZE_EXT" "D:\a\circt\circt\test\firtool\convention.fir"
$ ":" "RUN: at line 4"
$ "d:\a\circt\circt\build_release\bin\firtool.exe" "D:\a\circt\circt\test\firtool\convention.fir" "--format=fir" "--parse-only" "--scalarize-top-module=true" "--scalarize-ext-modules=true" "--preserve-aggregate=all"
note: command had no output on stdout or stderr
error: command failed with exit status: 0xc0000005
$ "d:\a\circt\circt\llvm\install\bin\filecheck.exe" "--check-prefix=SCALARIZE_BOTH" "D:\a\circt\circt\test\firtool\convention.fir"

--
```

It seems unlikely that commit broke things, we'll see if it continues to fail.  Weird that it fails this way, somewhat concerning!

teqdruid, 2023-05-12:
I re-ran and they passed https://github.com/llvm/circt/actions/runs/4963196181/jobs/8882164625

What I suspect is going on is that one or more of the tests are bumping up against the memory limit of the VM (which doesn't have much memory) so tests running at the same time as the memory hog fail. If I'm correct, the solution is to turn off parallel testing on the Windows CI builds.

teqdruid, 2023-05-12:
IIRC, we've seen stuff like this since the free VMs are puny.

closed 2023-05-25 by dtzSiFive