llvm/circt #5218: [FIRRTL] Tap annotation into external module should be changed to tap of ref port

darthscsi, 2023-05-17:
An external module with a data tap should represent that as an external module with a probe port and depend on the probe port ABI to reach into it.  This rather than verbatim strings floating around as suffixes to XMRRefOp.

seldridge, 2023-05-19:
Concretely, the following should compile and use the ABI:

```
circuit TestHarness:
  extmodule DUT:
    input clock: Clock
    output read: Probe<UInt<32>>
    output write: RWProbe<UInt<32>>

  module TestHarness:
    input clock: Clock

    inst dut of DUT
    dut.clock <= clock

    printf(clock, UInt<1>(1) "%x", read(dut.read))
    force_initial(dut.write, UInt<32>("hdeadbeef"))
```

Currently this will complain that the reference can't be traced.

dtzSiFive, 2023-05-19:
Oh! I thought this was about the use of `internalPath` for data taps, especially the `ref.send` of verbatim used sometimes.

Anyway, thanks for the concrete example + drawing my attention to this not being handled yet.  On it.