llvm/circt #5260: [Pipeline] representing multicycle/stage-skipping values

mortbopet, 2023-05-25:
Currently, all SSA def-use chains that cross a stage-separating operation will be registered, see [rationale](https://github.com/llvm/circt/blob/89e5c4c32947d512be29cf17dec5425cb8f2f1fe/docs/Dialects/Pipeline/RationalePipeline.md). We need a way to add:
1. latency semantics to a multi-cycle operation (most important)
2. pass in wires to an arbitrary point in the pipeline (less important, but related).

The design space here is... large... so i'd happily get your feedback and view of the problem.

## Example:

The following example includes a two-cycle adder, and a wire which needs to be available at all points in the pipeline without being registered.

```mlir
%out = pipeline.pipeline(%arg0, %arg1, %myWire %go) clock %clk reset %rst : (i32, i32, i32, i1) -> (i32) {
  ^bb0(%a0 : i32, %a1: i32, %myWire : %i32 %g : i1):
    // Say, %add0 has a 2 cycle latency
    %add0 = foo.slowAdder2 %a0, %a1 : i32

    // Say %myWire is a wire that we for some reason want to route to an arbitrary
    // point in the pipeline and NEVER want to register it, regardless of the scheduling
    // result, or if new pipeline.ss operations are added. In this toy result, we just
   // return it at the output of the pipeline.

    %s0_valid = pipeline.ss enable %g
    %s1_valid = pipeline.ss enable %s0_valid

    // %add0 becomes available here, and we want to stall it an extra cycle
    %s2_valid = pipeline.ss enable %s1_valid

    // Return %add0 which is delayed 1 cycle wrt. the origin operation
    pipeline.return %add0, %myWire valid %s2_valid : i32
}
```

Here's a few design points to solve the problem:

## Option 1: Attach latency semantics to the source operation

In this case, we would attach latency semantics as an attribute of the operation, denoting the latency of each result value of said operation.
The main drawback here is that this doesn't solve the wire problem, since we don't have an operation to attach to.

```mlir
%out = pipeline.pipeline(%arg0, %arg1, %myWire %go) clock %clk reset %rst : (i32, i32, i32, i1) -> (i32) {
  ^bb0(%a0 : i32, %a1: i32, %myWire : %i32 %g : i1):
    // Attach latency semantics to the operation as an attribute (res# : latency)
    %add0 = foo.slowAdder2 %a0, %a1 : i32 {pipeline.latency {0 : 2}}

    %s0_valid = pipeline.ss enable %g
    %s1_valid = pipeline.ss enable %s0_valid
    %s2_valid = pipeline.ss enable %s1_valid

    pipeline.return %add0 valid %s2_valid : i32
}
```

## Option 2: `passthrough` on stage delimiter ops

In this case, we don't have any latency semantics, but rather create the stage separating registers explicitly stating that a given SSA value should not be registered through that stage. 
An issue here is that adding or removing `pipeline.ss` ops will require updating the passthrough list, and
we have to explicitly _always_ add pass semantics for %myWire.

```mlir
%out = pipeline.pipeline(%arg0, %arg1, %myWire %go) clock %clk reset %rst : (i32, i32, i32, i1) -> (i32) {
  ^bb0(%a0 : i32, %a1: i32, %myWire : %i32 %g : i1):
    %add0 = foo.slowAdder2 %a0, %a1 : i32

    %s0_valid = pipeline.ss enable %g pass %add0, %myWire
    %s1_valid = pipeline.ss enable %s0_valid pass %add0, %myWire
    %s2_valid = pipeline.ss enable %s1_valid pass %myWire

    pipeline.return %add0, %myWire valid %s2_valid : i32
}
```

## Option 3: Pass operation

In this case, we introduce a new operation to the pipeline dialect, `pipeline.pass %value by $amt` with the semantics of "%value should pass the next $amt stages_". It is possible to specify a value for `$amt`, or `inf`, i.e. a wire available in the entire pipeline.

Why call it "pass" instead of "delay" or "latency"? Given the way that `pipeline` uses `pipeline.ss` in combination with SSA def-use chains across `pipeline.ss`-boundaries, thus making registers very much an implicit thing, i feel that calling something "delay" implies the registering of something wherein in reality it is the absence of it. Calling something "latency" would work for e.g. the adder operation, but it feels like a misnomer for the wire.  

A drawback of this is that nothing is preventing the use of the "original" value, i.e. the value that is provided as an input to the `pipeline.pass` operation.

```mlir
%out = pipeline.pipeline(%arg0, %arg1, %myWire %go) clock %clk reset %rst : (i32, i32, i32, i1) -> (i32) {
  ^bb0(%a0 : i32, %a1: i32, %myWire : %i32 %g : i1):
    %add0 = foo.slowAdder2 %a0, %a1 : i32

    // Add a pass operation, which indicates how many cycles a value is delayed
    // wrt. the stages that appear in front of it.
    %add0_pass = pipeline.pass %add0 by 2

    // Can add an "inf" value to indicate that the value must never be registered.
    %wire_pass = pipeline.pass %myWire by inf

    // 'pass' indicates that this stage shouldn't register %add0
    %s0_valid = pipeline.ss enable %g
    // %add0_pass is unavailable here and illegal to refence, can throw an error.
    %s1_valid = pipeline.ss enable %s0_valid
    // %add0_pass is available here
    %s2_valid = pipeline.ss enable %s1_valid

  
    // Drawback - we must make sure to reference the pass-returned SSA value instead
    // of the original SSA value.
    pipeline.return %add0_pass, %wire_pass valid %s2_valid : i32
}
```

## Option 4 separate "pass" top-level input for external wires, `pipeline.pass` op is a container.
We may add a separate input to the `pipeline.pipeline` operation which denotes "wire-like" inputs to the pipeline.
This would (for the sake of readability) be best implemented if we change the pipeline printer/parser to something like the following:

```
%out = pipeline.pipeline(%a0 = %arg0, %a1 = %arg1, %g = %go) wires (%innerWire : i32 = %myWire) clock %clk reset %rst : (i32, i32, i32, i1) -> (i32) {
    %add0 = pipeline.pass (%a0, %a1) : (i32, i32) -> i32 by 2 {
      ^bb0(%a0 : i32, %a1 : i32):
      %0 = foo.slowAdder2 %a0, %a1 : i32
      pipeline.pass.return %0 : i32
    }
    
    %s0_valid = pipeline.ss enable %g
    // %add0 is unavailable here and illegal to refence, can throw an error.
    %s1_valid = pipeline.ss enable %s0_valid
    // %add0_pass is available here
    %s2_valid = pipeline.ss enable %s1_valid

    pipeline.return %add0_pass, %innerWire valid %s2_valid : i32
}```

andrewb1999, 2023-05-25:
I think this is a slightly different issue than on the loop pipelining side of things. In loop pipeline we already represent the explicit start times of ops with the `loopschedule.stage` op, which seems to be what is missing here to enable stage-skipping. I agree that explicitly representing latencies is a good idea, so I think option 1 or something similar would be the best solution here.

jopperm, 2023-05-25:
I kind of like Option 4, because there is an integer explicitly telling the latency/number of stages to be skipped, and it prevents access to the "raw" value. This poking-holes-through-the-pipeline-stages-semantics would also make sense if such an op is not in the first stage IMHO.

Q: Could it be even simpler without the arguments, like this:

```mlir
%out = pipeline.pipeline(%arg0, %arg1, %myWire %go) clock %clk reset %rst : (i32, i32, i32, i1) -> (i32) {
  ^bb0(%a0 : i32, %a1: i32, %myWire : %i32 %g : i1):
    
    %add0 = pipeline.pass -> i32 by 2 {
      %0 = foo.slowAdder2 %a0, %a1 : i32
      pipeline.pass.return %0 : i32
    }

    %innerWire = pipeline.pass -> i32 by inf {
      pipeline.pass.return %myWire : i32
    }
    
    %s0_valid = pipeline.ss enable %g
    // %add0 is unavailable here and illegal to refence, can throw an error.
    %s1_valid = pipeline.ss enable %s0_valid
    // %add0 is available here
    %s2_valid = pipeline.ss enable %s1_valid

    pipeline.return %add0, %innerWire valid %s2_valid : i32
}
```

teqdruid, 2023-06-07:
Based on an internal discussion this morning, we agreed to go with option 4. I prefer @jopperm's syntax as well.

closed 2023-06-20 by mortbopet