llvm/circt #5273: Move operations to use MLIR Properties instead of Attributes

dtzSiFive, 2023-05-26:
This would be advantageous for many operations, and in many cases should be a simple conversion.

See:

https://mlir.llvm.org/OpenMeetings/2023-02-09-Properties.pdf

https://discourse.llvm.org/t/rfc-introducing-mlir-operation-properties/67846

And recent upstream MLIR commits migrating their dialects/IR over.

teqdruid, 2023-07-11:
for reference, https://reviews.llvm.org/D148299 moves the linalg dialect over using the dialect flag.

dtzSiFive, 2023-07-11:
Oh, glad you're interested/following this!  Thanks for the pointer, neat how they managed backwards compat!

I have a branch I've been sitting on that ports the HW-y dialects over (IIRC mostly just setting the flag, like upstream MLIR dialects did), but SV and FIRRTL will need some more careful handling/auditing before feel good about it.

And the benefit really starts once we move over and can start taking advantage of them-- just opting in in my testing only increased memory usage slightly (which makes sense, as I understand it, although didn't deep dive).

Happy to make a PR with that as a starting point if folks besides me are interested in this direction!

teqdruid, 2023-07-12:
I'm a big fan of properties in theory, and am very interested in moving things over. Less DictAttr building, more efficient get/set, though at the cost of more memory use in some scenarios.

If you're willing to push on it, I'd suggest moving one dialect at a time. I'll move the esi and msft dialects over when I get the chance.

dtzSiFive, 2023-08-30:
This is now the default, but we can opt-out until LLVM 19 by setting `let usePropertiesForAttributes = 0;`.  See: https://reviews.llvm.org/D158581 .

Nightly has started failing for (at least) this reason.

teqdruid, 2023-08-30:
> Nightly has started failing for (at least) this reason

I'm not seeing the failures. Can you point me to one?

teqdruid, 2023-08-30:
Oh... you mean the nightly track changes.