llvm/circt #5295: Flip polarity of CAPNP_DISABLE?

mikeurbach, 2023-05-30:
@teqdruid after basing the new cmake features on the CAPNP_DISABLE flag, anyone with an older cmake is going to have to set CAPNP_DISABLE=1, even if they have no idea what ESI cosim is or why they need to disable the capnp feature. Do you think it would make sense to flip the polarity, and default CAPNP_DISABLE to 1, or make it a flag like CAPNP_ENABLE? I'm personally having to go and set this flag now on all the environments with an older cmake, and I know others have as well.

teqdruid, 2023-05-30:
I'm fine with that. Won't have a chance to make the change myself until at
soonest tonight. If you need it sooner, feel free to do it.

As for disable vs enable: let's be consistent with what everything else is
doing.

mikeurbach, 2023-05-30:
Great, glad this makes sense in general. I am actually going on vacation myself but can take a look at switching this next week if you don't get a chance. I just wanted to bring it up before I left, since the goal was to not cause friction for people who don't need the features that require the latest cmake, but there is still some minor friction.

teqdruid, 2023-06-07:
wait... why don't we just move the cmake version logic into the CapnProto_FOUND block?

closed 2023-06-07 by teqdruid