llvm/circt #5296: How can I lower calyx dialect to hwdialect, I may have some question.

linuxlonelyeagle, 2023-05-31:
I am learning the calyx dialect.I know calyx can be lowered to fsm dialect by the picture of lower's route inside the official documentation.
```
// Here is my code.
calyx.component @main(%go: i1 {go}, %clk: i1 {clk}, %reset: i1 {reset}) -> (%done: i1 {done}) { 
  %r.in, %r.write_en, %r.clk, %r.reset, %r.out, %r.done = calyx.register @r : i1, i1, i1, i1, i1, i1
  %c1_1 = hw.constant 1 : i1
  calyx.wires {
   calyx.group @A {
      calyx.assign %r.in = %c1_1 : i1 
      calyx.assign %r.write_en = %c1_1 : i1
      calyx.group_done %r.done : i1
    }
  }
  
  calyx.control {
    calyx.seq {
        calyx.seq {
          calyx.enable @A
        }
    }
  }
}
```
I found a problem that I could not lower the calyx dialect to the fsm, which prevented me from lowering it further.
```
$ circt-opt calyx.mlir -lower-calyx-to-fsm

module {
  calyx.component @main(%go: i1 {go}, %clk: i1 {clk}, %reset: i1 {reset}) -> (%done: i1 {done}) {
    %r.in, %r.write_en, %r.clk, %r.reset, %r.out, %r.done = calyx.register @r : i1, i1, i1, i1, i1, i1
    %true = hw.constant true
    calyx.wires {
      calyx.group @A {
        calyx.assign %r.in = %true : i1
        calyx.assign %r.write_en = %true : i1
        calyx.group_done %r.done : i1
      }
    }
    calyx.control {
      fsm.machine @control() attributes {compiledGroups = [@A], initialState = "fsm_entry"} {
        fsm.state @fsm_entry output {
          fsm.output
        } transitions {
          fsm.transition @seq_0
        }
        fsm.state @fsm_exit output {
          fsm.output
        } transitions {
        }
        fsm.state @seq_0 output {
          fsm.output
        } transitions {
          fsm.transition @seq_0_seq_0_A
        }
        fsm.state @seq_0_seq_0_A output {
          calyx.enable @A
          fsm.output
        } transitions {
          fsm.transition @fsm_exit
        }
      }
    }
  }
}
```
Then I found the `lower-calyx-to-hw` pass, but it reported an error.Next I used -calyx-remove-groups, -calyx-remove-comb-groups, but it still didn't work.I hope someone can help me.Thanks!
```
circt-opt calyx.mlir -lower-calyx-to-fsm  -lower-calyx-to-hw
calyx.mlir:5:4: error: failed to legalize operation 'calyx.group' that was explicitly marked illegal
   calyx.group @A {
   ^
calyx.mlir:5:4: note: see current operation: 
"calyx.group"() ({
  "calyx.assign"(%21#0, %22) : (i1, i1) -> ()
  "calyx.assign"(%21#1, %22) : (i1, i1) -> ()
  "calyx.group_done"(%21#5) : (i1) -> ()
}) {sym_name = "A"} : () -> ()
```

rachitnigam, 2023-06-03:
@linuxlonelyeagle were you able to find the cause for this issue?

linuxlonelyeagle, 2023-06-03:
> @linuxlonelyeagle were you able to find the cause for this issue?

Sorry.I really didn't find the reason for this, I just wrote the code for calyx dialect after the [code](https://docs.calyxir.org/lang/multi-component.html) for the multi-component design of the calyx project.I am not very familiar with the circt project either, and I am trying to learn through this opportunity.

linuxlonelyeagle, 2023-06-03:
> @linuxlonelyeagle were you able to find the cause for this issue?

If I have time, I will try to find this issue.

linuxlonelyeagle, 2023-06-08:
> @linuxlonelyeagle were you able to find the cause for this issue?

I have successfully detected the bug. I expect to fix the bug and submit a patch.:smiley:

mikeurbach, 2023-06-08:
Closed by https://github.com/llvm/circt/pull/5341

closed 2023-06-08 by mikeurbach