llvm/circt #5309: [Arc][ArcCanonicalizer] Segfault

maerhart, 2023-06-02:
ArcCanonicalizer crashes on attached input. I was able to reduce the testcase to 600 lines of IR (reduction of 93% compared to original), but reducing it further seems to make it trigger only sometimes.

I would guess that the SymbolCache is not properly updated by one of the patterns, but don't actually know.

Stack trace on a RelWithDebugInfo build:
```
PLEASE submit a bug report to https://github.com/llvm/circt and include the crash backtrace.
Stack dump:
0.      Program arguments: build/bin/circt-opt -arc-canonicalizer input-reduced-prune2.mlir --mlir-timing
 #0 0x0000560c1915155e llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) /home/merhart/circt/llvm/llvm/lib/Support/Unix/Signals.inc:602:13
 #1 0x0000560c1914f925 llvm::sys::RunSignalHandlers() /home/merhart/circt/llvm/llvm/lib/Support/Signals.cpp:105:18
 #2 0x0000560c19151eac SignalHandler(int) /home/merhart/circt/llvm/llvm/lib/Support/Unix/Signals.inc:413:1
 #3 0x00007fa80f503520 (/lib/x86_64-linux-gnu/libc.so.6+0x42520)
 #4 0x0000560c191dbd32 mlir::detail::InterfaceMap::compare(mlir::TypeID, mlir::TypeID) /home/merhart/circt/llvm/mlir/include/mlir/Support/InterfaceSupport.h:250:37
 #5 0x0000560c191dbd32 auto mlir::detail::InterfaceMap::lookup(mlir::TypeID) const::'lambda'(auto const&, mlir::TypeID)::operator()<std::pair<mlir::TypeID, void*>>(auto const&, mlir::TypeID) const /home/merhart/circt/llvm/mlir/include/mlir/Support/InterfaceSupport.h:258:18
 #6 0x0000560c191dbd32 bool __gnu_cxx::__ops::_Iter_comp_val<mlir::detail::InterfaceMap::lookup(mlir::TypeID) const::'lambda'(auto const&, mlir::TypeID)>::operator()<std::pair<mlir::TypeID, void*> const*, mlir::TypeID const>(auto, mlir::TypeID const&) /usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/bits/predefined_ops.h:196:16
 #7 0x0000560c191dbd32 auto std::__lower_bound<std::pair<mlir::TypeID, void*> const*, mlir::TypeID, __gnu_cxx::__ops::_Iter_comp_val<mlir::detail::InterfaceMap::lookup(mlir::TypeID) const::'lambda'(auto const&, mlir::TypeID)>>(auto, auto, mlir::TypeID const&, __gnu_cxx::__ops::_Iter_comp_val<mlir::detail::InterfaceMap::lookup(mlir::TypeID) const::'lambda'(auto const&, mlir::TypeID)>) /usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/bits/stl_algobase.h:1464:8
 #8 0x0000560c191dbd32 auto std::lower_bound<std::pair<mlir::TypeID, void*> const*, mlir::TypeID, mlir::detail::InterfaceMap::lookup(mlir::TypeID) const::'lambda'(auto const&, mlir::TypeID)>(auto, auto, mlir::TypeID const&, mlir::detail::InterfaceMap::lookup(mlir::TypeID) const::'lambda'(auto const&, mlir::TypeID)) /usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/bits/stl_algo.h:2021:14
 #9 0x0000560c191dbd32 auto llvm::lower_bound<llvm::SmallVector<std::pair<mlir::TypeID, void*>, 3u> const&, mlir::TypeID&, mlir::detail::InterfaceMap::lookup(mlir::TypeID) const::'lambda'(auto const&, mlir::TypeID)>(auto&&, mlir::TypeID&, mlir::detail::InterfaceMap::lookup(mlir::TypeID) const::'lambda'(auto const&, mlir::TypeID)) /home/merhart/circt/llvm/llvm/include/llvm/ADT/STLExtras.h:2045:10
#10 0x0000560c191dbd32 mlir::detail::InterfaceMap::lookup(mlir::TypeID) const /home/merhart/circt/llvm/mlir/include/mlir/Support/InterfaceSupport.h:257:9
#11 0x0000560c191dbd32 circt::arc::CallOpMutableInterface::Concept* mlir::detail::InterfaceMap::lookup<circt::arc::CallOpMutableInterface>() const /home/merhart/circt/llvm/mlir/include/mlir/Support/InterfaceSupport.h:207:52
#12 0x0000560c191dbd32 circt::arc::CallOpMutableInterface::Concept* mlir::OperationName::getInterface<circt::arc::CallOpMutableInterface>() const /home/merhart/circt/llvm/mlir/include/mlir/IR/OperationSupport.h:344:41
#13 0x0000560c191dbd32 mlir::OpInterface<circt::arc::CallOpMutableInterface, circt::arc::detail::CallOpMutableInterfaceInterfaceTraits>::getInterfaceFor(mlir::Operation*) /home/merhart/circt/llvm/mlir/include/mlir/IR/OpDefinition.h:2068:34
#14 0x0000560c191dca8e mlir::detail::Interface<circt::arc::CallOpMutableInterface, mlir::Operation*, circt::arc::detail::CallOpMutableInterfaceInterfaceTraits, mlir::Op<circt::arc::CallOpMutableInterface>, mlir::OpTrait::TraitBase>::classof(mlir::Operation*) /home/merhart/circt/llvm/mlir/include/mlir/Support/InterfaceSupport.h:124:42
#15 0x0000560c191dca8e llvm::CastInfo<circt::arc::CallOpMutableInterface, mlir::Operation*, void>::isPossible(mlir::Operation*) /home/merhart/circt/llvm/mlir/include/mlir/IR/Operation.h:1047:58
#16 0x0000560c191dca8e llvm::ConstStrippingForwardingCast<circt::arc::CallOpMutableInterface, mlir::Operation const*, llvm::CastInfo<circt::arc::CallOpMutableInterface, mlir::Operation*, void>>::isPossible(mlir::Operation const* const&) /home/merhart/circt/llvm/llvm/include/llvm/Support/Casting.h:396:12
#17 0x0000560c191dca8e llvm::CastInfo<circt::arc::CallOpMutableInterface, mlir::Operation* const, void>::isPossible(mlir::Operation* const&) /home/merhart/circt/llvm/llvm/include/llvm/Support/Casting.h:509:12
#18 0x0000560c191dca8e bool llvm::isa<circt::arc::CallOpMutableInterface, mlir::Operation*>(mlir::Operation* const&) /home/merhart/circt/llvm/llvm/include/llvm/Support/Casting.h:549:10
#19 0x0000560c191dca8e auto (anonymous namespace)::SinkArcInputsPattern::matchAndRewrite(circt::arc::DefineOp, mlir::PatternRewriter&) const::$_5::operator()<mlir::Operation>(mlir::Operation*) const /home/merhart/circt/lib/Dialect/Arc/Transforms/ArcCanonicalizer.cpp:421:43
#20 0x0000560c191dca8e bool __gnu_cxx::__ops::_Iter_pred<(anonymous namespace)::SinkArcInputsPattern::matchAndRewrite(circt::arc::DefineOp, mlir::PatternRewriter&) const::$_5>::operator()<mlir::Operation* const*>(mlir::Operation* const*) /usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/bits/predefined_ops.h:318:16
#21 0x0000560c191dca8e mlir::Operation* const* std::__find_if<mlir::Operation* const*, __gnu_cxx::__ops::_Iter_pred<(anonymous namespace)::SinkArcInputsPattern::matchAndRewrite(circt::arc::DefineOp, mlir::PatternRewriter&) const::$_5>>(mlir::Operation* const*, mlir::Operation* const*, __gnu_cxx::__ops::_Iter_pred<(anonymous namespace)::SinkArcInputsPattern::matchAndRewrite(circt::arc::DefineOp, mlir::PatternRewriter&) const::$_5>, std::random_access_iterator_tag) /usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/bits/stl_algobase.h:2099:8
#22 0x0000560c191dca8e mlir::Operation* const* std::__find_if<mlir::Operation* const*, __gnu_cxx::__ops::_Iter_pred<(anonymous namespace)::SinkArcInputsPattern::matchAndRewrite(circt::arc::DefineOp, mlir::PatternRewriter&) const::$_5>>(mlir::Operation* const*, mlir::Operation* const*, __gnu_cxx::__ops::_Iter_pred<(anonymous namespace)::SinkArcInputsPattern::matchAndRewrite(circt::arc::DefineOp, mlir::PatternRewriter&) const::$_5>) /usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/bits/stl_algobase.h:2114:14
#23 0x0000560c191dca8e mlir::Operation* const* std::find_if<mlir::Operation* const*, (anonymous namespace)::SinkArcInputsPattern::matchAndRewrite(circt::arc::DefineOp, mlir::PatternRewriter&) const::$_5>(mlir::Operation* const*, mlir::Operation* const*, (anonymous namespace)::SinkArcInputsPattern::matchAndRewrite(circt::arc::DefineOp, mlir::PatternRewriter&) const::$_5) /usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/bits/stl_algo.h:3910:14
#24 0x0000560c191dca8e bool std::none_of<mlir::Operation* const*, (anonymous namespace)::SinkArcInputsPattern::matchAndRewrite(circt::arc::DefineOp, mlir::PatternRewriter&) const::$_5>(mlir::Operation* const*, mlir::Operation* const*, (anonymous namespace)::SinkArcInputsPattern::matchAndRewrite(circt::arc::DefineOp, mlir::PatternRewriter&) const::$_5) /usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/bits/stl_algo.h:471:24
#25 0x0000560c191dca8e bool std::any_of<mlir::Operation* const*, (anonymous namespace)::SinkArcInputsPattern::matchAndRewrite(circt::arc::DefineOp, mlir::PatternRewriter&) const::$_5>(mlir::Operation* const*, mlir::Operation* const*, (anonymous namespace)::SinkArcInputsPattern::matchAndRewrite(circt::arc::DefineOp, mlir::PatternRewriter&) const::$_5) /usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/bits/stl_algo.h:490:15
#26 0x0000560c191dca8e bool llvm::any_of<llvm::ArrayRef<mlir::Operation*>&, (anonymous namespace)::SinkArcInputsPattern::matchAndRewrite(circt::arc::DefineOp, mlir::PatternRewriter&) const::$_5>(llvm::ArrayRef<mlir::Operation*>&, (anonymous namespace)::SinkArcInputsPattern::matchAndRewrite(circt::arc::DefineOp,  [... cut ...]
#61 0x0000560c191149a6 main /home/merhart/circt/tools/circt-opt/circt-opt.cpp:71:23
#62 0x00007fa80f4ead90 __libc_start_call_main ./csu/../sysdeps/nptl/libc_start_call_main.h:58:16
#63 0x00007fa80f4eae40 call_init ./csu/../csu/libc-start.c:128:20
#64 0x00007fa80f4eae40 __libc_start_main ./csu/../csu/libc-start.c:379:5
#65 0x0000560c19114405 _start (build/bin/circt-opt+0xea405)
```
[arccanonicalizer-crash-reduced.mlir.gz](https://github.com/llvm/circt/files/11634288/arccanonicalizer-crash-reduced.mlir.gz)

hovind, 2024-01-12:
Do you happen to have an updated test case lying around, @maerhart? The provided test case IR is in an outdated format: `arccanonicalizer-crash-reduced.mlir:404:25: error: custom op 'hw.module' expected port direction`.

maerhart, 2024-01-12:
This should be the same IR in the new format: [input-reduced-prune2.mlir.gz](https://github.com/llvm/circt/files/13922344/input-reduced-prune2.mlir.gz)

I suspect that this segfault already got fixed in [a recent commit](https://github.com/llvm/circt/commit/e76a757e067f239ccff9bc62421337ce55982a22#diff-9c691711ba146085f29309d7c115eb048836a340f75d47984e5019a217d0a3ba) because I had a segault that looked very similar (maybe the same?) there as well and introduced the `ArcListener` to fix it. I did a quick check now and couldn't reproduce it. Let me know if you can't observe it anymore as well, then I'll just close the issue.

hovind, 2024-01-14:
> This should be the same IR in the new format: [input-reduced-prune2.mlir.gz](https://github.com/llvm/circt/files/13922344/input-reduced-prune2.mlir.gz)
>
> I suspect that this segfault already got [...]

Thanks for the updated test case!

I can't observe the segfault either. Sure enough, I'm able to reproduce the `SEGFAULT` on ff7271b2ec09b6e289d4332b6d34b075dd6522a0, but not on the following commit, e76a757e067f239ccff9bc62421337ce55982a22. Well done!

maerhart, 2024-01-14:
That's very cool to hear, thanks for double-checking!

closed 2024-01-14 by maerhart