llvm/circt #5324: getFieldName (and friends?) support for enums

dtzSiFive, 2023-06-06:
Following input crashes due to getFieldName not being supported (used to report error in InferWidths):

```firrtl
circuit NoWidthEnum:
  module NoWidthEnum:
    output o : {| Some : UInt, None |}
```

youngar, 2023-06-21:
https://github.com/llvm/circt/pull/5439 fixes `getFieldName`.  I was looking at other helpers we are missing support for, and notably we are missing support in `getValueByFieldID`, which is used for GrandCentral and WireDFT.  I don't think that it makes sense for these passes to target a specific variant of an EnumerationType, it should probably be limited to targeting the enumeration as a whole.  We should probably enforce that limitation in how we attach annotations.

youngar, 2023-06-21:
Opened an issue for the above comment here: https://github.com/llvm/circt/issues/5441

PR #5439 (2023-06-21): [FIRRTL] Support EnumTypes in error messages
This adds support for enumeration types to `getFieldName`, which is commonly used to print the access path to a field in an aggregate.  This function is used to report error messages in InferWidths.

Fixes #5324
commit 30a7e7a6 (2023-06-21): [FIRRTL] Support EnumTypes in error messages (#5439)

This adds support for enumeration types to `getFieldName`, which is
commonly used to print the access path to a field in an aggregate.  This
function is used to report error messages in InferWidths.

Fixes #5324
closed 2023-06-21 by youngar