llvm/circt #5329: [LLHD][FIRRTL] Valgrind errors in nightly tests

teqdruid, 2023-06-07:
I could have sworn there was an issue for this, but the nightly tests have been failing for some time now. Valgrind is failing on the normal tests, not even getting to the integration tests.

```
********************
Failed Tests (13):
  CIRCT :: Dialect/FIRRTL/parse-basic.fir
  CIRCT :: Dialect/LLHD/Simulator/sim_arbitrary_signal_size.mlir
  CIRCT :: Dialect/LLHD/Simulator/sim_bit_precision_drives.mlir
  CIRCT :: Dialect/LLHD/Simulator/sim_comb_concat.mlir
  CIRCT :: Dialect/LLHD/Simulator/sim_formats.mlir
  CIRCT :: Dialect/LLHD/Simulator/sim_hw_array_get_concat.mlir
  CIRCT :: Dialect/LLHD/Simulator/sim_hw_bitcast.mlir
  CIRCT :: Dialect/LLHD/Simulator/sim_hw_struct_extract_inject.mlir
  CIRCT :: Dialect/LLHD/Simulator/sim_process.mlir
  CIRCT :: Dialect/LLHD/Simulator/sim_reg.mlir
  CIRCT :: Dialect/LLHD/Simulator/sim_shifts.mlir
  CIRCT :: Dialect/LLHD/Simulator/sim_simple.mlir
  CIRCT :: Dialect/LLHD/Simulator/sim_wait.mlir
```

https://github.com/llvm/circt/actions/runs/5188247337/jobs/9351636792

dtzSiFive, 2023-06-07:
( #5253 , but yes still failing :disappointed: )

teqdruid, 2023-06-07:
How did I miss that?

closed 2023-06-07 by teqdruid