llvm/circt #5336: [OM][Python] Windows build failure

teqdruid, 2023-06-08:
```
[4605/5039] Building CXX object tools\circt\lib\Bindings\Python\CMakeFiles\CIRCTPythonModules.extension._circt.dso.dir\OMModule.cpp.obj
FAILED: tools/circt/lib/Bindings/Python/CMakeFiles/CIRCTPythonModules.extension._circt.dso.dir/OMModule.cpp.obj 
C:\BuildTools\VC\Tools\MSVC\14.29.30133\bin\Hostx64\x64\cl.exe  /nologo /TP -DCIRCTPythonModules_extension__circt_dso_EXPORTS -DMLIR_PYTHON_PACKAGE_PREFIX=circt. -DUNICODE -D_CRT_NONSTDC_NO_DEPRECATE -D_CRT_NONSTDC_NO_WARNINGS -D_CRT_SECURE_NO_DEPRECATE -D_CRT_SECURE_NO_WARNINGS -D_GLIBCXX_ASSERTIONS -D_LIBCPP_ENABLE_ASSERTIONS -D_SCL_SECURE_NO_DEPRECATE -D_SCL_SECURE_NO_WARNINGS -D_UNICODE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -IC:\__w\1\b\build\tools\circt\lib\Bindings\Python -IC:\__w\1\s\CIRCT\circt\lib\Bindings\Python -IC:\__w\1\b\build\include -IC:\__w\1\s\CIRCT\circt\llvm\llvm\include -IC:\__w\1\s\CIRCT\circt\include -IC:\__w\1\b\build\tools\circt\include -external:IC:\__w\1\s\CIRCT\circt\llvm\llvm\..\mlir\include -external:IC:\__w\1\b\build\tools\mlir\include -external:I"C:\Program Files\Python311\include" -external:I"C:\Program Files\Python311\Lib\site-packages\pybind11\include" -external:W0 /DWIN32 /D_WINDOWS   /Zc:inline /Zc:preprocessor /Zc:__cplusplus /Oi /bigobj /permissive- /W4 -wd4141 -wd4146 -wd4244 -wd4267 -wd4291 -wd4351 -wd4456 -wd4457 -wd4458 -wd4459 -wd4503 -wd4624 -wd4722 -wd4100 -wd4127 -wd4512 -wd4505 -wd4610 -wd4510 -wd4702 -wd4245 -wd4706 -wd4310 -wd4701 -wd4703 -wd4389 -wd4611 -wd4805 -wd4204 -wd4577 -wd4091 -wd4592 -wd4319 -wd4709 -wd5105 -wd4324 -w14062 -we4238 /Gw /MD /O2 /Ob2 -std:c++17 -UNDEBUG /EHsc /GR /bigobj /MP /GL /showIncludes /Fotools\circt\lib\Bindings\Python\CMakeFiles\CIRCTPythonModules.extension._circt.dso.dir\OMModule.cpp.obj /Fdtools\circt\lib\Bindings\Python\CMakeFiles\CIRCTPythonModules.extension._circt.dso.dir\ /FS -c C:\__w\1\s\CIRCT\circt\lib\Bindings\Python\OMModule.cpp
C:\__w\1\s\CIRCT\circt\lib\Bindings\Python\OMModule.cpp(63): error C2039: 'base': is not a member of 'std::_Vector_iterator<std::_Vector_val<std::_Simple_types<_Ty>>>'
        with
        [
            _Ty=MlirAttribute
        ]
C:\BuildTools\VC\Tools\MSVC\14.29.30133\include\vector(457): note: see declaration of 'std::_Vector_iterator<std::_Vector_val<std::_Simple_types<_Ty>>>'
        with
        [
            _Ty=MlirAttribute
        ]
C:\__w\1\s\CIRCT\circt\lib\Bindings\Python\OMModule.cpp(62): error C2660: 'omEvaluatorInstantiate': function does not take 3 arguments
C:\__w\1\s\CIRCT\circt\include\circt-c/Dialect/OM.h(80): note: see declaration of 'omEvaluatorInstantiate'
```

Unfortunately, CIRCT doesn't (yet) have the integration tests running on Windows so this sort of thing slips through. We run a pipeline internally to do this.

teqdruid, 2023-06-08:
Hold please. That's a build from two weeks ago. Running a fresh one from today.

teqdruid, 2023-06-08:
Yeah, same problem with 71fd2dc84e5481a38565e0fead357362dbd5da99

mikeurbach, 2023-06-08:
Hmm it seems similar to this for example: https://stackoverflow.com/questions/21643668/microsoft-visualstudio-2010-c-vector-iterator-base-not-found. Apparently this part of the stdlib is implementation defined or something that I didn't realize. I will try to fix it, but I don't have access to a windows machine. Maybe docker can work for testing?

mikeurbach, 2023-06-08:
I think the linked PR might fix it.

closed 2023-06-08 by mikeurbach

teqdruid, 2023-06-09:
Seems to have fixed the problem. Ran into #5346 while verifying the fix, but since subsequent ninja runs passed, I think this fix worked. Thanks for fixing it so quick!