llvm/circt #5337: Why python bindings need a unified build?

ColdCodeCool, 2023-06-08:
As the title shows, why Circt needs a unified build? Can I build llvm first and then build circt?

mikeurbach, 2023-06-08:
This is a historical artifact from when MLIR was first enabling Python bindings, and at that time it was how the various projects in the community committed to doing it, which we discussed and committed to for CIRCT: https://discourse.llvm.org/t/rfc-unified-cmake-builds-for-python-at-least/3965.

I believe that it may be possible to get Python bindings working with a pre-built LLVM, but I'm not sure where the support has ended up there. For the reasons mentioned in the above thread, there's really no downside to using a unified build. Our README doesn't recommend it, but frankly I think it should.

ColdCodeCool, 2023-06-08:
@mikeurbach thank you, Mike. I build my own project, but I came up to a problem that I cannot build a custom attribute in python. But I did register it in pybind like circt did. 
LLVM ERROR: can't create Attribute 'xxxxxAttr' because storage uniquer isn't initialized: the dialect was likely not loaded, or the attribute wasn't added with addAttributes<...>() in the Dialect::initialize() method.

mikeurbach, 2023-06-08:
Hmm, I can't recall the exact failure mode(s) that happen when the Python projects are not using a unified build. This may or may not be related to a unified build.

Is this attribute something you've defined in a fork of CIRCT? Or your own dialect? I'm trying to understand how this relates to CIRCT.

Regardless, I think we will need to see more about exactly how you've declared the attribute and registered the dialect in order to help. In CIRCT, we define this function: https://github.com/llvm/circt/blob/0a67d92405ba2f89654c13e476c378fec90ebbe6/lib/Bindings/Python/CIRCTModule.cpp#L50

And you must call that on a context before you can use any of the dialects: https://github.com/llvm/circt/blob/0a67d92405ba2f89654c13e476c378fec90ebbe6/integration_test/Bindings/Python/dialects/comb.py#L11

ColdCodeCool, 2023-06-09:
> Hmm, I can't recall the exact failure mode(s) that happen when the Python projects are not using a unified build. This may or may not be related to a unified build.
>
> Is this attribute something [...]

Thank you, Mike. I define them in my own projects, not in circt. Thanks for the answer, I will try it out.

ColdCodeCool, 2023-07-14:
Hi @mikeurbach , I succeeded to build a custom attribute, but I cannot print it correctly. I post the question in LLVM discourse
https://discourse.llvm.org/t/custom-attribute-printing-hook-in-python-bindings/72031. Do I need to repost here? can you take a look or I can issue another thread.

mikeurbach, 2023-07-14:
Interesting, I will try to answer on discourse.