llvm/circt #5346: [OM] Missing CMake dependency (?)

teqdruid, 2023-06-09:
Discovered in Windows build, but I'm pretty sure it's not Windows specific. Ninja build.

New problem:

```
[3396/5086] Building CXX object tools\circt\lib\Dialect\OM\Evaluator\CMakeFiles\obj.CIRCTOMEvaluator.dir\Evaluator.cpp.obj
FAILED: tools/circt/lib/Dialect/OM/Evaluator/CMakeFiles/obj.CIRCTOMEvaluator.dir/Evaluator.cpp.obj 
C:\BuildTools\VC\Tools\MSVC\14.29.30133\bin\Hostx64\x64\cl.exe  /nologo /TP -DGTEST_HAS_RTTI=0 -DUNICODE -D_CRT_NONSTDC_NO_DEPRECATE -D_CRT_NONSTDC_NO_WARNINGS -D_CRT_SECURE_NO_DEPRECATE -D_CRT_SECURE_NO_WARNINGS -D_GLIBCXX_ASSERTIONS -D_HAS_EXCEPTIONS=0 -D_LIBCPP_ENABLE_ASSERTIONS -D_SCL_SECURE_NO_DEPRECATE -D_SCL_SECURE_NO_WARNINGS -D_UNICODE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -IC:\__w\1\b\build\tools\circt\lib\Dialect\OM\Evaluator -IC:\__w\1\s\CIRCT\circt\lib\Dialect\OM\Evaluator -IC:\__w\1\b\build\include -IC:\__w\1\s\CIRCT\circt\llvm\llvm\include -IC:\__w\1\s\CIRCT\circt\include -IC:\__w\1\b\build\tools\circt\include -external:IC:\__w\1\s\CIRCT\circt\llvm\llvm\..\mlir\include -external:IC:\__w\1\b\build\tools\mlir\include -external:W0 /DWIN32 /D_WINDOWS   /Zc:inline /Zc:preprocessor /Zc:__cplusplus /Oi /bigobj /permissive- /W4 -wd4141 -wd4146 -wd4244 -wd4267 -wd4291 -wd4351 -wd4456 -wd4457 -wd4458 -wd4459 -wd4503 -wd4624 -wd4722 -wd4100 -wd4127 -wd4512 -wd4505 -wd4610 -wd4510 -wd4702 -wd4245 -wd4706 -wd4310 -wd4701 -wd4703 -wd4389 -wd4611 -wd4805 -wd4204 -wd4577 -wd4091 -wd4592 -wd4319 -wd4709 -wd5105 -wd4324 -w14062 -we4238 /Gw /MD /O2 /Ob2 -std:c++17  /EHs-c- /GR- -UNDEBUG /showIncludes /Fotools\circt\lib\Dialect\OM\Evaluator\CMakeFiles\obj.CIRCTOMEvaluator.dir\Evaluator.cpp.obj /Fdtools\circt\lib\Dialect\OM\Evaluator\CMakeFiles\obj.CIRCTOMEvaluator.dir\ /FS -c C:\__w\1\s\CIRCT\circt\lib\Dialect\OM\Evaluator\Evaluator.cpp
C:\__w\1\s\CIRCT\circt\include\circt/Dialect/OM/OMTypes.h(20): fatal error C1083: Cannot open include file: 'circt/Dialect/OM/OMTypes.h.inc': No such file or directory
```

Subsequent runs didn't have that error, which indicates a missing CMake dep. (Which I have not been able to find.)

teqdruid, 2023-06-09:
Do we need the following? I don't see any other libs doing this.

```
diff --git a/lib/Dialect/OM/Evaluator/CMakeLists.txt b/lib/Dialect/OM/Evaluator/CMakeLists.txt
index 2df5351ec..9c28fad5e 100644
--- a/lib/Dialect/OM/Evaluator/CMakeLists.txt
+++ b/lib/Dialect/OM/Evaluator/CMakeLists.txt
@@ -1,6 +1,10 @@
 add_circt_library(CIRCTOMEvaluator
   Evaluator.cpp

+  DEPENDS
+  MLIROMIncGen
+  MLIROMAttrIncGen
+
   LINK_LIBS
   CIRCTOM
   MLIRIR
```

mikeurbach, 2023-06-09:
Hmm I saw this in a single run of many when I was testing a wheel building change as well. I will take a look, those extra DEPENDS shouldn't be needed and the CIRCTOM library should already depend on them correctly.

mikeurbach, 2023-06-09:
Actually, we probably do need to add those dependencies, I will add them. We do depend on that generated file from the Evaluator library, so I guess it must be declared directly instead of transitively to be safe.

teqdruid, 2023-06-09:
I'm thinking link_libs doesn't wait for the lib to actually be built before
starting the compile of the depending lib.

closed 2023-06-09 by mikeurbach