llvm/circt #5360: [FIRRTL] Useless wire names could be useful as debug information

uenoku, 2023-06-12:
https://github.com/llvm/circt/pull/5288 changed DropName pass to literally drop useless names (mostly created as Chisel temporaries) with default options for better verilog output but these names were previously used for error messages. For example, ExpandWhen is not able to produce good messages for wires created as Chisel temporary. 
```scala
circuit Foo :
  module Foo :
    wire _foo_WIRE: UInt<1> @[foo.scala 0:0]
```

```sh
$ firtool foo.fir
foo.scala:0:0: error: sink "" not fully initialized in module "Foo"
```

darthscsi, 2023-06-12:
Isn't `--preserved-values=named` what we are using?  In this case it looks good.

closed 2023-06-12 by darthscsi

dtzSiFive, 2023-06-12:
Issue maybe should be filed re:producing better diagnostics when names aren't found/present, specifically in ExpandWhen's but would be good to look around for where else this might be prone to producing things like `""`.

commit e8aa6366 (2023-06-12): [FIRRTL] Change _ names to demote in preserve mode none

Closes #5360