llvm/circt #5420: [Python] Wheel upload doesn't work on MacOS

mikeurbach, 2023-06-16:
@rsetaluri It appears the step we are using to publish wheels to PyPI doesn't work on MacOS: https://github.com/llvm/circt/actions/runs/5293588613/jobs/9581906975#step:8:16.

I believe the recommended workflow in this case is to run all the jobs, with each of them publishing artifacts to a shared place, and a final job that runs on linux and does the upload: https://github.com/pypa/gh-action-pypi-publish/discussions/15

mikeurbach, 2023-06-16:
For now I've turned off fail-fast (the default): https://github.com/llvm/circt/commit/b63bca1e624aea3eaf17d3f3a1d6614415cfc4cb. This should allow Linux builds to publish while we sort this out.

rsetaluri, 2023-07-19:
#5628 should fix this issue. Keeping an eye on nightly builds to ensure that the fix is correct.

rsetaluri, 2023-07-20:
For some reason it looks like the depending job (`Push wheels`) didn't get triggered. Wonder if it's the wrong `if` condition: https://github.com/llvm/circt/actions/runs/5610739087.

rsetaluri, 2023-07-21:
> For some reason it looks like the depending job (`Push wheels`) didn't get triggered. Wonder if it's the wrong `if` condition: https://github.com/llvm/circt/actions/runs/5610739087.

Ok, my guess is that if **any** of the jobs fail, as [Build wheels with cp37-macosx_x86_64](https://github.com/llvm/circt/actions/runs/5610739087/job/15210803603#logs) did in the above run), then the depending job won't run at all. So I think if we fix #5634 everything will be happy :)

mikeurbach, 2023-07-21:
> Ok, my guess is that if any of the jobs fail

Hmm yeah maybe we can tweak the action. I've already set `fail-fast: false` so if one row of the matrix fails, the others continue. Maybe there is a similar setting for the final push job, to not require all previous jobs to succeed.

leonardt, 2023-07-21:
I may be out of date, but last I spoke to @rsetaluri he said you were pursuing a path where you upload artifacts and then download to a single job to do the upload.

if it's helpful to have an example, here's a relatively simple workflow that does that https://github.com/Kuree/uinspect/blob/master/.github/workflows/ci.yml

rsetaluri, 2023-07-22:
Thanks @leonardt, think we have the solution for this; #5628 seems to do the trick.

closed 2023-07-22 by rsetaluri

leonardt, 2023-08-10:
Is there a reason we don't release a wheel for Python 3.9? (e.g. was there some issue?) Otherwise should be simple to add another entry.

mikeurbach, 2023-08-10:
I don't think there is any specific reason not to have Python 3.9, most likely it's just the [usual](https://mlir.llvm.org/getting_started/Faq/#why-is-dialect-x-missing-feature-y) explanation that no one has needed it.

leonardt, 2023-08-10:
I snuck adding 3.9 into https://github.com/llvm/circt/pull/5822