llvm/circt #5424: Illegal assignment pattern [IAP] error when using VCS-2014 for simulation on RocketConfig (from Chipyard)

fangrouli, 2023-06-19:
(Originally posted at Chipyard, being advised to here.)

Bug report

**Chipyard Version and Hash**
Release: 1.9.0
Hash: 7475bfb1a05802ac4dfc3990a889f93164b8d798

**OS Setup**
System 1:
Linux ... 5.19.0-45-generic #46~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC ... 20 x86_64 GNU/Linux
Using Conda to set up, conda-lock = 1.4.0

System 2:
Linux ... 2.6.32-754.el6.x86_64 #1 SMP ... 2018 x86_64 x86_64 x86_64 GNU/Linux
Using Synopsys 2014.09

**Other Setup for Reproduce**
The .sv files are generated on the Ubuntu 22.04 system by command `make verilog` in `sims/vcs/` directory. The generated files in `chipyard.TestHarness.RocketConfig` is copied to another terminal with CentOS 6 for simulation. The CentOS 6 system has GCC and G++ version of 4.7.2

**Current Behavior**
Generating the verilog for the default `RocketConfig` and copy to the VCS-located terminal.
Source the synopsys source file.
Running the [Makefile](https://drive.google.com/file/d/16crR9sJ7Itlq77SX4e8C0skc4PdPEL_A/view?usp=sharing) with command `make run`.
This is the [Error log](https://drive.google.com/file/d/1F0q87DsmvyhaAJs-P0WxFtrYqINokL8Y/view?usp=drive_link).

In summary, VCS compiler detect a lists of illegal assignment pattern in various .sv files generated, for example in `Aomics.sv`:
```
wire [3:0][3:0] _GEN = '{4'hC, 4'h8, 4'hE, 4'h6};	// @[Atomics.scala:42:8]
```
will give the IAP error.

**Expected Behavior**
No error, successful compilation with a simv executable generated.

**Other Information**
I have already included systemverilog option in VCS command, so really not sure what is the cause of the error. Please point the error cause to me, thank you very much!

uenoku, 2023-06-20:
I guess old VCS doesn't accept the array literal for packed arrays. The fix would be to remove `'` here:
https://github.com/llvm/circt/blob/dcf8d1b0bd45e63cc51ae0eb30c3d0478b5b9768/lib/Conversion/ExportVerilog/ExportVerilog.cpp#L2584

We can create a lowering option for that, but perhaps we can just remove `'` globally.

fangrouli, 2023-06-20:
Then will setting `disallowPackedArrays` help? Because I install firtool with a tar file instead of clonnig the git repo, so not very sure whether I can find where to make the mentioned change locally. Thank you so much!

uenoku, 2023-06-21:
`disallowPackedArrays` has a known issue (like https://github.com/llvm/circt/issues/5355) so maybe it doesn't work. It it compiles with `disallowPackedArrays`, then the issue above should be fixed. Either way, I fixed the problematic emission on ToT so please use a newer version of firtool.

closed 2023-06-21 by uenoku