llvm/circt #5436: [OM][Python][Windows]

teqdruid, 2023-06-20:
```
FAILED: tools/circt/python_packages/circt_core/circt/_mlir_libs/_circt.cp311-win_amd64.pyd 
cmd.exe /C "cd . && "C:\Program Files\CMake\bin\cmake.exe" -E vs_link_dll --intdir=tools\circt\lib\Bindings\Python\CMakeFiles\CIRCTPythonModules.extension._circt.dso.dir --rc="C:\Program Files (x86)\Windows Kits\10\bin\10.0.19041.0\x64\rc.exe" --mt="C:\Program Files (x86)\Windows Kits\10\bin\10.0.19041.0\x64\mt.exe" --manifests  -- C:\BuildTools\VC\Tools\MSVC\14.29.30133\bin\Hostx64\x64\link.exe /nologo tools\circt\lib\Bindings\Python\CMakeFiles\CIRCTPythonModules.extension._circt.dso.dir\CIRCTModule.cpp.obj tools\circt\lib\Bindings\Python\CMakeFiles\CIRCTPythonModules.extension._circt.dso.dir\ESIModule.cpp.obj tools\circt\lib\Bindings\Python\CMakeFiles\CIRCTPythonModules.extension._circt.dso.dir\HWModule.cpp.obj tools\circt\lib\Bindings\Python\CMakeFiles\CIRCTPythonModules.extension._circt.dso.dir\OMModule.cpp.obj tools\circt\lib\Bindings\Python\CMakeFiles\CIRCTPythonModules.extension._circt.dso.dir\MSFTModule.cpp.obj tools\circt\lib\Bindings\Python\CMakeFiles\CIRCTPythonModules.extension._circt.dso.dir\SVModule.cpp.obj  /out:tools\circt\python_packages\circt_core\circt\_mlir_libs\_circt.cp311-win_amd64.pyd /implib:tools\circt\python_packages\circt_core\circt\_mlir_libs\_circt.lib /pdb:tools\circt\python_packages\circt_core\circt\_mlir_libs\_circt.pdb /dll /version:0.0 /machine:x64 /INCREMENTAL:NO  -LTCG  tools\circt\python_packages\circt_core\circt\_mlir_libs\CIRCTBindingsPythonCAPI.lib  "C:\Program Files\Python311\libs\python311.lib"  "C:\Program Files\Python311\libs\python311.lib"  lib\LLVMSupport.lib  psapi.lib  shell32.lib  ole32.lib  uuid.lib  advapi32.lib  delayimp.lib  -delayload:shell32.dll  -delayload:ole32.dll  lib\LLVMDemangle.lib  kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib  && cd ."
LINK: command "C:\BuildTools\VC\Tools\MSVC\14.29.30133\bin\Hostx64\x64\link.exe /nologo tools\circt\lib\Bindings\Python\CMakeFiles\CIRCTPythonModules.extension._circt.dso.dir\CIRCTModule.cpp.obj tools\circt\lib\Bindings\Python\CMakeFiles\CIRCTPythonModules.extension._circt.dso.dir\ESIModule.cpp.obj tools\circt\lib\Bindings\Python\CMakeFiles\CIRCTPythonModules.extension._circt.dso.dir\HWModule.cpp.obj tools\circt\lib\Bindings\Python\CMakeFiles\CIRCTPythonModules.extension._circt.dso.dir\OMModule.cpp.obj tools\circt\lib\Bindings\Python\CMakeFiles\CIRCTPythonModules.extension._circt.dso.dir\MSFTModule.cpp.obj tools\circt\lib\Bindings\Python\CMakeFiles\CIRCTPythonModules.extension._circt.dso.dir\SVModule.cpp.obj /out:tools\circt\python_packages\circt_core\circt\_mlir_libs\_circt.cp311-win_amd64.pyd /implib:tools\circt\python_packages\circt_core\circt\_mlir_libs\_circt.lib /pdb:tools\circt\python_packages\circt_core\circt\_mlir_libs\_circt.pdb /dll /version:0.0 /machine:x64 /INCREMENTAL:NO -LTCG tools\circt\python_packages\circt_core\circt\_mlir_libs\CIRCTBindingsPythonCAPI.lib C:\Program Files\Python311\libs\python311.lib C:\Program Files\Python311\libs\python311.lib lib\LLVMSupport.lib psapi.lib shell32.lib ole32.lib uuid.lib advapi32.lib delayimp.lib -delayload:shell32.dll -delayload:ole32.dll lib\LLVMDemangle.lib kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib /MANIFEST /MANIFESTFILE:tools\circt\python_packages\circt_core\circt\_mlir_libs\_circt.cp311-win_amd64.pyd.manifest" failed (exit code 1120) with the following output:
   Creating library tools\circt\python_packages\circt_core\circt\_mlir_libs\_circt.lib and object tools\circt\python_packages\circt_core\circt\_mlir_libs\_circt.exp
OMModule.cpp.obj : error LNK2001: unresolved external symbol "public: class llvm::StringRef __cdecl mlir::StringAttr::getValue(void)const " (?getValue@StringAttr@mlir@@QEBA?AVStringRef@llvm@@XZ)
OMModule.cpp.obj : error LNK2001: unresolved external symbol "public: class llvm::ArrayRef<class mlir::Attribute> __cdecl mlir::ArrayAttr::getValue(void)const " (?getValue@ArrayAttr@mlir@@QEBA?AV?$ArrayRef@VAttribute@mlir@@@llvm@@XZ)
OMModule.cpp.obj : error LNK2001: unresolved external symbol "private: static class mlir::SelfOwningTypeID mlir::detail::TypeIDResolver<class mlir::ArrayAttr,void>::id" (?id@?$TypeIDResolver@VArrayAttr@mlir@@X@detail@mlir@@0VSelfOwningTypeID@3@A)
OMModule.cpp.obj : error LNK2001: unresolved external symbol "private: static class mlir::SelfOwningTypeID mlir::detail::TypeIDResolver<class mlir::StringAttr,void>::id" (?id@?$TypeIDResolver@VStringAttr@mlir@@X@detail@mlir@@0VSelfOwningTypeID@3@A)
tools\circt\python_packages\circt_core\circt\_mlir_libs\_circt.cp311-win_amd64.pyd : fatal error LNK1120: 4 unresolved externals
```

The Windows python build is broken. OMModule is missing a dependency.

teqdruid, 2023-06-20:
I'm guessing it's missing `MLIRIR`. Building on local Windows now to repro and confirm.

mikeurbach, 2023-06-20:
I think probably I should rewrite this to use the MLIR C APIs, which it already depends on. I'm not sure if we are still in the world where we're not supposed to use C++ APIs in our pybind11 code, but there are already C APIs I can use for this so that will be an easy fix.

teqdruid, 2023-06-20:
Simply adding MLIRIR to the deps fixes the linking problem... Though I agree that the pybind11 code shouldn't be using MLIR constructs. That's the whole reason we go through the CAPI!

mikeurbach, 2023-06-20:
Yes, thanks for fixing it but I think this is hacky. I'm about to send a patch that should fix it by using the actual MLIR CAPI, which should already be depended on correctly.

teqdruid, 2023-06-20:
I agree. Will close my PR.

mikeurbach, 2023-06-21:
This should fix that: https://github.com/llvm/circt/commit/80531ebc4f13d4b0f589567a5a0c18aa0bc0b02e. If I'm understanding how `add_mlir_public_c_api_library` and `add_mlir_python_common_capi_library`, we shouldn't need any additional dependencies, since `CIRCTBindingsPythonCAPI` should already aggregate the dependencies of `CIRCTCAPIOM`, which does depend on `MLIRCAPIIR`.

teqdruid, 2023-06-21:
That helped a bit... now I'm running into

```
$ ninja -C ..\circt_build_2019.nocapnp\ -j 16 check-circt-integration
ninja: Entering directory `..\circt_build_2019.nocapnp\'
[15/24] Building CXX object tools\circt\lib\Bindings\Python\CMakeFiles\CIRCTPythonModules.extension._circt.dso.dir\OMModule.cpp.obj
FAILED: tools/circt/lib/Bindings/Python/CMakeFiles/CIRCTPythonModules.extension._circt.dso.dir/OMModule.cpp.obj
C:\PROGRA~2\MIB055~1\2019\ENTERP~1\VC\Tools\MSVC\1429~1.301\bin\Hostx64\x64\cl.exe  /nologo /TP -DCIRCTPythonModules_extension__circt_dso_EXPORTS -DMLIR_PYTHON_PACKAGE_PREFIX=circt. -DUNICODE -D_CRT_NONSTDC_NO_DEPRECATE -D_CRT_NONSTDC_NO_WARNINGS -D_CRT_SECURE_NO_DEPRECATE -D_CRT_SECURE_NO_WARNINGS -D_GLIBCXX_ASSERTIONS -D_LIBCPP_ENABLE_ASSERTIONS -D_SCL_SECURE_NO_DEPRECATE -D_SCL_SECURE_NO_WARNINGS -D_UNICODE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -IC:\code\circt_build_2019.nocapnp\tools\circt\lib\Bindings\Python -IC:\code\circt\lib\Bindings\Python -IC:\code\circt_build_2019.nocapnp\include -IC:\code\circt\llvm\llvm\include -IC:\code\circt\include -IC:\code\circt_build_2019.nocapnp\tools\circt\include -external:IC:\code\circt\llvm\llvm\..\mlir\include -external:IC:\code\circt_build_2019.nocapnp\tools\mlir\include -external:IC:\Python\Python39\include -external:IC:\Python\Python39\Lib\site-packages\pybind11\include -external:W0 /DWIN32 /D_WINDOWS   /Zc:inline /Zc:preprocessor /Zc:__cplusplus /Oi /bigobj /permissive- /W4 -wd4141 -wd4146 -wd4244 -wd4267 -wd4291 -wd4351 -wd4456 -wd4457 -wd4458 -wd4459 -wd4503 -wd4624 -wd4722 -wd4100 -wd4127 -wd4512 -wd4505 -wd4610 -wd4510 -wd4702 -wd4245 -wd4706 -wd4310 -wd4701 -wd4703 -wd4389 -wd4611 -wd4805 -wd4204 -wd4577 -wd4091 -wd4592 -wd4319 -wd4709 -wd5105 -wd4324 -w14062 -we4238 /Gw /MD /O2 /Ob2 -std:c++17 -UNDEBUG /EHsc /GR /bigobj /MP /GL /showIncludes /Fotools\circt\lib\Bindings\Python\CMakeFiles\CIRCTPythonModules.extension._circt.dso.dir\OMModule.cpp.obj /Fdtools\circt\lib\Bindings\Python\CMakeFiles\CIRCTPythonModules.extension._circt.dso.dir\ /FS -c C:\code\circt\lib\Bindings\Python\OMModule.cpp
C:\Python\Python39\include\pyconfig.h(59): fatal error C1083: Cannot open include file: 'io.h': No such file or directory
[22/24] Running utility command for CIRCTPythonModules.sources.MLIRPythonSources.Core
```

Though I don't think it's related and is probably just a local build issue. *Sigh* Windows.

mikeurbach, 2023-06-21:
Hmm ok, that `io.h` business seems unrelated to MLIR dependencies. If you think there's still something off about the OM bindings, please let me know.

teqdruid, 2023-06-21:
Yep. Just what I figured. I'm a moron.

Your fix did indeed work. Thanks for the quick fix!

closed 2023-06-21 by teqdruid