llvm/circt #5442: [FIRRTL][CheckCombLoops] Non-deterministic ability to find a combinational cycle

youngar, 2023-06-21:
The following example has a combinational loop, but sometimes it is not found by the pass.  Running `firtool` on this example:
```
circuit Foo:

  module Bar:
    input a : UInt<1>
    output b : UInt<1>
    b <= a

  module Baz:
    input a : UInt<1>
    output b : UInt<1>
    output c : {d : UInt<1>}
    b <= a
    c.d <= a

  module Foo:
    inst bar of Bar
    inst baz of Baz

    bar.a <= baz.b
    baz.a <= bar.b
```
Will sometimes produce no error, or sometimes produce:
```
./test.fir:15:10: error: detected combinational cycle in a FIRRTL module, sample path: Foo.{bar.a <- baz.b <- baz.a <- bar.b <- bar.a}
  module Foo:
```

I suspect that the non-deterministic iteration order of `handlePorts` mixed with the early return might be responsible for this issue: 
https://github.com/llvm/circt/blob/30a7e7a6f18dcaf598640eb432fa4264f2927ad2/lib/Dialect/FIRRTL/Transforms/CheckCombLoops.cpp#L386-L396

prithayan, 2023-06-21:
Thanks a lot @youngar for filing the issue, and also the fix for it. This was a nasty bug, nothing would be working correctly because of it. Fortunately/unfortunately, this bug was just resulting in missed cycle detection, which is why it survived for so long. 
I committed the fix directly to main by mistake, instead of a PR. Will wait for the checks to pass, then close it.

dtzSiFive, 2023-06-22:
Fixed in https://github.com/llvm/circt/commit/67abd8617e0d5debcc7012aac0299244405fe437, test from issue report added in following commit: https://github.com/llvm/circt/commit/f7d3c706cec84dd5f8a87de316ff44d99b46cf10 .

Looks good, thanks for the report and thanks for the quick fix!

closed 2023-06-22 by dtzSiFive