llvm/circt #5444: [FIRRTL][InferWidths] Back-prop for mux selector operand; different than subaccess

dtzSiFive, 2023-06-21:
Before getting into details:

The FIRRTL spec does say the selector will be inferred to width `1`, so this may be "working as intended" / spec detail to be resolved.  It is inconsistent with subaccess, as noted below.

Onwards to the issue, consider:

```firrtl
circuit Mux:
  module Mux:
    input x : UInt<0>
    output y : UInt
    output z : UInt

    wire w : UInt
    w <= x

    y <= w

    wire buffer : UInt
    buffer <= w
    z <= mux(buffer, UInt(1), UInt(2))
```
vs

```firrtl
circuit Mux:
  module Mux:
    input x : UInt<0>
    output y : UInt
    output z : UInt

    wire w : UInt
    w <= x

    y <= w

    ; wire buffer : UInt
    ; buffer <= w
    z <= mux(w, UInt(1), UInt(2))
```
 
Which produces the following outputs (1.34.0 to avoid #5391):
```systemverilog
// Generated by CIRCT firtool-1.34.0
module Mux(
  output [1:0] z
);

  assign z = 2'h2;
endmodule
```
and
```systemverilog
module Mux(
  output       y,
  output [1:0] z
);

  assign y = 1'h0;
  assign z = 2'h2;
endmodule
```

Note that the presence of the wire `buffer` (which could be optimized/elided for various reasons) changes the widths and presence of hardware on different path.

Probably we don't want any back-propagation at all (other than `attach`), or should rework how widths are inferred in spec (perhaps to something more unification-oriented + solving together, vs minimum to make input connections legal / forward-only).

----

This behavior does not happen for the "equivalent" subaccess operation, which is a bit inconsistent (thanks to @seldridge/@uenoku for pointing this out).

dtzSiFive, 2023-06-27:
(Looks like SFC 1.5.3 produces equivalent outputs for the two designs, FWIW.)

jackkoenig, 2023-06-27:
Yeah this is definitely a bit weird

> Probably we don't want any back-propagation at all (other than attach), or should rework how widths are inferred in spec (perhaps to something more unification-oriented + solving together, vs minimum to make input connections legal / forward-only).

I agree. I think we should get rid of this back-propagation because it is inconsistent with everything else. We can consider a more unification-style algorithm, but we obviously need to consider it holistically and it'd be a pretty big change across the board [I think at least].

In any case, detecting such cases and warning probably isn't too difficult. Users who like the semantics can just explicitly set the width of the component doing the select to 1-bit, it's easy enough to deal with. I also expect this comes up very rarely, but I'd be curious to see the results of deploying a warning here.

azidar, 2023-07-05:
Yeah, this is weird, and I remember way back when confusing myself about whether the predicate should infer to width 1 or not. Have you considered (now that zero-width wires are supported) that we could back-propagate at least a 0 width wire (instead of a 1-width wire)? I believe this would fix this specific issue.

dtzSiFive, 2023-07-05:
Thanks the input/context, folks!

## Back-propagation

Partly this issue is regarding "why do we (only) do back-propagation / adding constraints to input for mux(-like) operations?".  It's unclear this is desired, or useful (how might a reasonable design rely on this that we want to preserve and isn't better served by either rejecting or deleting the relevant hardware?).  If there are use-cases for this that changes things, of course.

To be clear re:"back-propagation", the way this is modeled presently is equivalent to if we connected to the selector source (imagine was a wire) from a known-1-bit signal.  This means the signal can be entirely unconstrained otherwise and by virtue of being used in this way will get a width.

### Problem

This is a problem as it may cause unneeded hardware to be preserved, is inconsistent with other FIRRTL constructs, and breaks equivalence between designs using mux and other otherwise-equivalent constructs.  This ties semantics to require inspecting users of a value, and makes certain seemingly straightforward/desired transformations invalid:
* can't drop a dead mux as it may be what's keeping the design from being rejected
* can't fold away a useless mux for similar reason (`mux(c, x, x)`)
* introducing a mux changes width inference behavior, replacing it does as well, must be conservative if mux selector has unknown width

To expand on last point a bit--
As-is this difference compared to alternative formulations of the same means that when uninferred widths are involved, mux is not equivalent to almost any other construct and transformations need to be careful in this regard (either never converting or if okay to accept a design we would normally reject ("are width inference errors not a feature we must preserve") then could move in the more permissive direction (to mux but not other way?))...

Happily `when` is required to take a 1-bit condition so this doesn't happen for mux operations introduced that way[1].

Any changes to this behavior (back-propagation of any variety) will cause designs that previously worked to be rejected and addressed in the spec (where we can sort out how to gate/deploy/migrate).  It would be interesting to push some designs through a build that did not behave this way (Solution 2).

### Solution 1: Back-propagate zero-width

> Have you considered (now that zero-width wires are supported) that we could back-propagate at least a 0 width wire (instead of a 1-width wire)? I believe this would fix this specific issue.

As for avoiding keeping around unneeded hardware (as in the example), reworking this to instead treat this as a connection from a zero-bit wire (preventing considering these designs under-constrained) should avoid inferring upstream signals too wide.  Would want to check selector doesn't end up too large either (not handled well presently).
As a small note, for legalization we'd need to support multiplex operations w/smaller-width selectors, or special-handle these operations after updating in InferWidths-- something like extending the selector signal to the full bit width where width inference computed a selector of too narrow width (FWIW new mux intrinsic has 2-bit selector with same problem).

Anyway in short, that sort of thing could be part of addressing this (maybe best immediate solution, thanks!) but still presents a source of inconsistency and invalidates transformations that we probably want to be able to do (and probably do already).

### Solution 2: Don't back-propagate?

I wonder what would happen if we didn't back-propagate the width here, and just said it's more like connecting /to/ a 1-bit (or 2-bit) signal, extending as required (but if selector is unconstrained then reject)?
("What does back-propagation get us?" (that's worth the inconsistencies/special-handling))

Thinking a bit more, this concerns designs like this one:

```
circuit Issue5444Reg:
  module Issue5444Reg:
    input clock : Clock
    output out : UInt

    reg r1 : UInt, clock
    r1 <= mux(r1, r1, r1)
    out <= r1
```

(We could either infer to zero-width, or reject as unconstrained, or present behavior of infer to 1-bit)

-----

## Misc

[1] Looks like for SFC, this same issue can be triggered via uninferred `when` condition:
```
circuit Mux:
  module Mux:
    input x : UInt<0>
    output y : UInt
    output z : UInt

    wire w : UInt
    w <= x

    y <= w

    when w:
      z <= UInt(1)
    else :
      z <= UInt(2)
```

which produces:

```
module Mux(
  output       y,
  output [1:0] z
);
  assign y = 1'h0;
  assign z = 2'h2; // @[]
endmodule
```

(CIRCT produces error if the condition to `when` is not known to be 1-bit)

dtzSiFive, 2023-07-05:
("Solution 1" implemented as #5538)

azidar, 2023-07-05:
Thanks for the writeup!! After consideration, I think my vote is to not do width back propagation, but just reject the case that the predicate width is unresolved. E.g. solution 2.

PR #6917 (2024-04-17): [FIRRTL] no back-prop for width of mux selectors, support narrower
Support parsing mux with selector of zero-width, per spec.

Fix use as mux selector "back-propagating" (acts as-if connected to from a 1-bit source).

Canonicalize narrower selectors to full width for lowering and analysis simplicity.

Fixes #5444.
closed 2024-04-17 by dtzSiFive