llvm/circt #5454: [Seq] What is the canonical way to clock gate a register?

mortbopet, 2023-06-22:


teqdruid, 2023-06-23:
The advantage of have a clock enable is it'd be easy to map it into an FPGA primitive (which have a CE input). If we represent the notion of a CE by putting something between the reg and the clk signal, that makes clock domain analysis -- as well as any query along the lines of "what elements does this clock signal drive?" -- more complex.

My view is that `firreg` + `clock_gate` is strictly for ASICs. The assumption I'd like to make is that `compreg`'s `clk` input is the 'unmolested' clock. (Kinda part of the reason I was arguing years ago -- before the `seq` dialect -- for a `hw.clk` type.) If one wants to use `compreg.ce` in an ASIC design, it could be pretty trivially lowered to `firreg` + `clock_gate`. Actually, it'd probably be a good idea to lower `compreg` to `firreg` for ASIC designs also.

mortbopet, 2023-06-23:
That makes sense - might be good to change the seq rationale with that information... and rename `firreg` to something more general if we see that as being the canonical way to represent ASIC-targeting registers.