llvm/circt #5462: [FIRRTL][CheckCombCycles] Missed cycles, sensitive to presence of node?

dtzSiFive, 2023-06-22:
Input, `--preserve-aggregate=all`:

```firrtl
circuit Loop:
  module Loop:
    wire w : {a: UInt<8>}
    node n = w
    w <= n
```

For some reason the node is needed for this to be missed (?).

This was reduced from slightly more complicated interprocedural inputs like:

```firrtl
circuit Loop:
  module Child:
    input bundle : { a : UInt<1>, b : UInt<1> }
    output p : {a : UInt<1>, b : UInt<1> }

    node n = bundle

    p <= n
  module Loop:
    input x : UInt<1>

    inst c of Child
    c.bundle.a <= x
    c.bundle.b <= c.p.b
```

----

Does not appear to be limited to cycles entirely unreachable/disconnected from the rest of the design/ports, FWIW, consider:

```firrtl
circuit Loop:
  module Loop:
    input in : {a: UInt<8>}
    output out : {a: UInt<8>}
    input c : UInt<1>

    wire w : {a: UInt<8>}
    node n = w
    w <= mux(c, n, in)

    out <= w
```

dtzSiFive, 2023-06-22:
Note the last one causes firtool (canonicalizer?) to infinite loop in HW, probably due to the self-connect this becomes (?).

prithayan, 2023-06-23:
Thanks Will for the reduced testcase, I'm working on this.

seldridge, 2024-02-10:
Fixed with https://github.com/llvm/circt/commit/7032ca61b932e029fd3052d61e77a504b86681f3.

closed 2024-02-10 by seldridge