llvm/circt #5463: [Seq] Unify `seq`-dialect memory representations

mortbopet, 2023-06-23:
As noted in https://github.com/llvm/circt/pull/5009, https://github.com/llvm/circt/pull/5025, `seq.firmem` was introduced to initially help factoring out memory handling in the FIRRTL-to-HW pass. Once done, it is then intended that it and `seq.hlmem` can be unified into a single representation - `seq.mem` - seeing as they both essentially try to solve the same problem.

i.e.
* `seq.firmem` + `seq.hlmem` => `seq.mem`
* `seq.firmem.read_port/write_port/read_write_port` + `seq.read/write/read_write/ => `seq.read_port/write_port/read_write_port`

mortbopet, 2023-06-23:
CC @fabianschuiki

fabianschuiki, 2023-06-23:
Thanks for opening a tracking issue!

teqdruid, 2023-07-25:
@fabianschuiki How's this going?

fabianschuiki, 2023-07-25:
Still letting the dust settle on `seq.firmem` internally. This has been enabled in the last release of firtool, and I'm collecting feedback and looking out for issues that come up with it in the next few release cycles -- just in case we need to quickly revert it. After that I'm all for merging these two representations, and homogenizing the latency annotations as needed.

teqdruid, 2024-06-21:
@fabianschuiki certainly the dust has settled by now...

fabianschuiki, 2024-06-27:
Yeah we should definitely unify these 😿