llvm/circt #5479: [FIRRTL][IMDCE] Constant prop not working with const types

youngar, 2023-06-26:
IMDCE contains a simple constant prop, which appears to not be working with the addition of `const` types.   cc @trilorez @uenoku.

`./bin/circt-opt -pass-pipeline='builtin.module(firrtl-imdeadcodeelim)' ./test2.mlir --mlir-print-ir-after-all`

```mlir
firrtl.circuit "Test" {
  firrtl.module private @Blah(out %out : !firrtl.const.uint<1>) {
    %c1_ui1 = firrtl.constant 1 : !firrtl.const.uint<1>
    firrtl.strictconnect %out, %c1_ui1 : !firrtl.const.uint<1>
  }
  firrtl.module @Test(out %out : !firrtl.const.uint<1>) {
    %blah_out = firrtl.instance blah interesting_name @Blah(out out : !firrtl.const.uint<1>)
    firrtl.strictconnect %out, %blah_out : !firrtl.const.uint<1>
  }
}
```

```mlir
./test2.mlir:8:5: error: 'firrtl.strictconnect' op requires all operands to have the same type
    firrtl.strictconnect %out, %blah_out : !firrtl.const.uint<1>
    ^
./test2.mlir:8:5: note: see current operation: "firrtl.strictconnect"(%arg0, %0) : (!firrtl.const.uint<1>, !firrtl.uint<1>) -> ()
// -----// IR Dump After IMDeadCodeElim Failed (firrtl-imdeadcodeelim) //----- //
#loc = loc("./test2.mlir":6:27)
"builtin.module"() ({
  "firrtl.circuit"() ({
    "firrtl.module"() ({
    ^bb0(%arg0: !firrtl.const.uint<1>):
      %0 = "firrtl.constant"() {value = 1 : ui1} : () -> !firrtl.uint<1>
      "firrtl.strictconnect"(%arg0, %0) : (!firrtl.const.uint<1>, !firrtl.uint<1>) -> ()
    }) {annotations = [], convention = #firrtl<convention internal>, parameters = [], portAnnotations = [], portDirections = true, portLocations = [#loc], portNames = ["out"], portSyms = [], portTypes = [!firrtl.const.uint<1>], sym_name = "Test"} : () -> ()
  }) {annotations = [], name = "Test"} : () -> ()
}) : () -> ()
```

trilorez, 2023-06-26:
Where is this issue coming up? Const should be dropped from all types well before the IMDCE pass

trilorez, 2023-06-26:
Cool. For future reference, I've been keeping the old `dev/trilorez/firrtl-const-fir` branch around which has my initial const implementation that keeps const throughout the full pipeline. Here's how I handled IMDCE: https://github.com/llvm/circt/compare/dev/trilorez/firrtl-const-fir#diff-151e2a7ae19a8e08017e8c73414b4c8bfbd3e3288051904276ffe9f2fb150587