llvm/circt #5483: [ExportVerilog] Missing `hw.struct_explode` support?

7FM, 2023-06-26:
Running `circt-opt <file.mlir> --export-verilog` with `file.mlir` having the following content:
```
module {
  hw.module @EXPLODE(%0: i64, %1: i64) -> (arg1: i64, arg2: i64) {
    %2 = hw.struct_create (%0, %1) : !hw.struct<arg1: i64, arg2: i64>
    %arg1, %arg2 = hw.struct_explode %2 : !hw.struct<arg1: i64, arg2: i64>

    hw.output %arg1, %arg2 : i64, i64
  }
}
```
triggers the following assertion:
```
circt-opt: circt/lib/Conversion/ExportVerilog/ExportVerilog.cpp:3295:
void (anonymous namespace)::NameCollector::collectNames(mlir::Block &): 
Assertion `(!isExpr || isExpressionEmittedInline(&op, moduleEmitter.state.options))
 && "If 'op' is a verilog expression, the expression must be inlinable. " 
 "Otherwise, it is a bug of PrepareForEmission"' failed.
```

This seems to be strongly related to the `hw.struct_explode` operation, because replacing it with `hw.struct_extract` ops works just fine.

(Yes, I know that in this case, the canonicalizer would optimize `hw.struct_create` and `hw.struct_explode` away. But this is the smallest example that I could come up with to reproduce the bug.)

uenoku, 2023-06-27:
It looks like a bug of PrepareForEmission. `struct_explode` is legalized into multiple `struct_extract`s but uses of the input is not cloned properly.

hovind, 2024-04-26:
Seems to me like this has been fixed:
```
$ cat /tmp/explode.mlir 
module {
  hw.module @EXPLODE(in %0: i64, in %1: i64, out arg1: i64, out arg2: i64) {
    %2 = hw.struct_create (%0, %1) : !hw.struct<arg1: i64, arg2: i64>
    %arg1, %arg2 = hw.struct_explode %2 : !hw.struct<arg1: i64, arg2: i64>

    hw.output %arg1, %arg2 : i64, i64
  }
}
$ ./build/bin/circt-opt --export-verilog /tmp/explode.mlir 
// Generated by CIRCT unknown git version
module EXPLODE(	// build/bin/explode.mlir:2:3
  input  [63:0] _GEN,	// build/bin/explode.mlir:2:25
                _GEN_0,	// build/bin/explode.mlir:2:37
  output [63:0] arg1,	// build/bin/explode.mlir:2:50
                arg2	// build/bin/explode.mlir:2:65
);

  wire struct packed {logic [63:0] arg1; logic [63:0] arg2; } _GEN_1 =
    '{arg1: _GEN, arg2: _GEN_0};	// build/bin/explode.mlir:3:10
  assign arg1 = _GEN_1.arg1;	// build/bin/explode.mlir:3:10, :4:20, :6:5
  assign arg2 = _GEN_1.arg2;	// build/bin/explode.mlir:3:10, :4:20, :6:5
endmodule

module {
  hw.module @EXPLODE(in %0 "" : i64 {hw.verilogName = "_GEN"}, in %1 "" : i64 {hw.verilogName = "_GEN_0"}, out arg1 : i64, out arg2 : i64) {
    %2 = hw.struct_create (%0, %1) : !hw.struct<arg1: i64, arg2: i64>
    %3 = sv.wire {hw.verilogName = "_GEN_1"} : !hw.inout<struct<arg1: i64, arg2: i64>>
    sv.assign %3, %2 : !hw.struct<arg1: i64, arg2: i64>
    %4 = sv.read_inout %3 : !hw.inout<struct<arg1: i64, arg2: i64>>
    %arg1 = hw.struct_extract %4["arg1"] : !hw.struct<arg1: i64, arg2: i64>
    %arg2 = hw.struct_extract %4["arg2"] : !hw.struct<arg1: i64, arg2: i64>
    hw.output %arg1, %arg2 : i64, i64
  }
}
```

7FM, 2024-04-29:
Thanks for testing, I will close this issue then.