llvm/circt #5484: CIRCT cannot consume the generic format it generates

Moxinilian, 2023-06-26:
`circt-opt` cannot take as input the output it emitted using `-mlir-print-op-generic`. 

The reason this happens seems to be because of the location attributes attached to HwModule. Are they truly necessary? It seems to fail because they are inconsistent with block arguments. But is this check necessary? And if the locations can be found from the block arguments, are the explicit locations necessary?

## Instructions to reproduce

Give the following snippet to `circt-opt -mlir-print-op-generic`:

```mlir
hw.module @B(%a: i1) -> (nameOfPortInSV: i1, "": i1) {
  %0 = comb.or %a, %a : i1
  %1 = comb.and %a, %a : i1
  hw.output %0, %1: i1, i1
}
```

This should yield something like:
```
#loc = loc("<stdin>":1:14)
#loc1 = loc("<stdin>":1:26)
#loc2 = loc("<stdin>":1:46)
"builtin.module"() ({
  "hw.module"() ({
  ^bb0(%arg0: i1):
    %0 = "comb.or"(%arg0, %arg0) : (i1, i1) -> i1
    %1 = "comb.and"(%arg0, %arg0) : (i1, i1) -> i1
    "hw.output"(%0, %1) : (i1, i1) -> ()
  }) {argLocs = [#loc], argNames = ["a"], comment = "", function_type = (i1) -> (i1, i1), parameters = [], resultLocs = [#loc1, #loc2], resultNames = ["nameOfPortInSV", ""], sym_name = "B"} : () -> ()
}) : () -> ()
```

Giving this output to `circt-opt` again yields an error:
```
<stdin>:5:3: error: 'hw.module' op block argument locations should match signature locations
  "hw.module"() ({
  ^
```

maerhart, 2023-06-27:
Seems like redundant information to me that could be removed, but might lead to inconsistencies w.r.t the result locs since they are not controlled via the debug info CLI argument. Just removing the check, on the other hand, could lead to confusions on which loc to use for the args (e.g., in transformation passes). Best solution IMO would be to remove `arcLocs`, and make `resultLocs` emission dependent on the debuginfo CLI args, but that's probably not possible yet and would require changes to upstream. If it is emitted using `--mlir-print-debuginfo --mlir-print-op-generic` it should work, right?

The explicit port locations were added in #4540 by @youngar. Maybe he can provide more details on that?

Moxinilian, 2023-06-27:
I was not aware of `mlir-print-debuginfo`, thank you! That will at least solve my personal problem for now.
But it would be nice to not need it.

youngar, 2023-06-28:
I don't really have anything to add. This was definitely an unfortunate affect of the PR.  I was able to write the custom asm formatter to do the correct thing by querying `flags.shouldPrintDebugInfo()`, but there is no way to tell the generic printer to skip printing an attribute.

The array of debug information was added for things like external modules which don't have a region (and thus have no block arguments to attach a location to). To keep the code paths all similar, we decided to put the array on all types of modules.

It would help if regular modules did not replicate the information in two places, we could only store it on block arguments. Another option would be to stop adding port location information to ext-modules, it is of dubious usefulness in the first place.

teqdruid, 2024-02-01:
Is there any plan to resolve this? It's _super_ annoying.

closed 2024-02-14 by youngar