llvm/circt #5506: [FIRRTL] Bundle-of-alias crashes

dtzSiFive, 2023-06-28:
```
circuit AliasBundle:
  type Foo = { a : UInt<1> }
  type Bar = { a : Foo }
  module AliasBundle:
     input b : Bar
```

Which crashes presently, in InferWidths, in debug builds, reporting unsupported type encountered (Alias).

uenoku, 2023-06-28:
Yep, thank you for reporting. The migration (llvm::cast -> firrtl::type_cast) is not completed yet.

uenoku, 2023-08-07:
It now compiles. for example with `firtool bar.fir -preserve-aggregate=all -scalarize-top-module=false`:
```verilog
// Generated by CIRCT firtool-1.49.0-50-gbf3fc36b4-dirty
`ifndef _TYPESCOPE_AliasBundle__TYPESCOPE_
`define _TYPESCOPE_AliasBundle__TYPESCOPE_
  typedef struct packed {logic a; } Foo;
  typedef struct packed {Foo a; } Bar;
`endif // _TYPESCOPE_AliasBundle__TYPESCOPE_
module AliasBundle(
  input Bar b
);

endmodule
```

closed 2023-08-07 by uenoku