llvm/circt #5521: [SCF-to-Calyx] Improper ordering of `scf.while` and operators when lowering

calebmkim, 2023-06-29:
Hi! I ran into a bug lowering SCF to Calyx that I'll explain here. 

## Minimal Buggy Example 
(For the complete example, see: [ex-bug.mlir.txt](https://github.com/llvm/circt/files/11907996/ex-bug.mlir.txt) 
and run `circt-opt ex-bug.mlir --lower-scf-to-calyx`). But here's the gist: 

This SCF code: 
```
  // 1) load memory 
  %0 = memref.load %alloca[%c0] : memref<38xi32>
  // 2) while loop 
  %1 = scf.while (%arg0 = %c0) : (index) -> index {
    // some condition 
  } do {
    // some body 
  }
```
gets lowered to: 
```
calyx.seq {
  // this par + while correspond to the scf.while 
  calyx.par {
   // initialize register for while loop 
  }
  calyx.while %std_slt_0.out with @bb0_1 {
    // while body 
  }
  // this enable corresponds to the memref.load. This is the wrong order
  calyx.enable @bb0_0
}
```
As you can see, when we lower the design into Calyx, the while loop gets moved to before the `memref.load`, when it should have remained after. 
In general, it seems that while loops will always get moved in front of other operations when they're part of the same basic block. 

## What is causing this problem (I think) 
I want to emphasize that I'm new to the codebase, so I'm not sure if this is right. But I figured if it makes sense to y'all then this may be the problem, and it doesn't hurt to give more information. 

I think the problem is coming from the order in which the [patterns](https://github.com/llvm/circt/blob/006f7db42b078b1f9f5fe46572c482963b1a3922/lib/Conversion/SCFToCalyx/SCFToCalyx.cpp#L1344) are being applied. 
It seems that the `BuildWhileGroup` [pattern](https://github.com/llvm/circt/blob/006f7db42b078b1f9f5fe46572c482963b1a3922/lib/Conversion/SCFToCalyx/SCFToCalyx.cpp#L1344) is applied before the `BuildOpGroups` pattern. This is mostly fine. However, this [line](https://github.com/llvm/circt/blob/006f7db42b078b1f9f5fe46572c482963b1a3922/lib/Conversion/SCFToCalyx/SCFToCalyx.cpp#L912) in the `BuildWhileGroup` pattern adds the `WhileScheduleable` to the [SchedulerInterface](https://github.com/llvm/circt/blob/006f7db42b078b1f9f5fe46572c482963b1a3922/include/circt/Dialect/Calyx/CalyxLoweringUtils.h#L159C7-L159C25). Then, the `BuildOpGroup` pattern (which seems to be applied after `BuildWhileGroup`) also adds its own Schedulable to the same Scheduler Interface [later on](https://github.com/llvm/circt/blob/006f7db42b078b1f9f5fe46572c482963b1a3922/lib/Conversion/SCFToCalyx/SCFToCalyx.cpp#L252). This is where the problem arises: because the `BuildWhileGroup` pattern was applied first, the while loop is always going to appear first in the `SchedulerInterface` (if they're within the same block). Then, when the `BuildControl` pattern looks thru the SchedulerInterface [here](https://github.com/llvm/circt/blob/006f7db42b078b1f9f5fe46572c482963b1a3922/lib/Conversion/SCFToCalyx/SCFToCalyx.cpp#L950), it will see the while loop first (because it was added first), and therefore spit out the calyx while loop before the other operations. 

## Solution
I can try to do some hacking on this to see if I can solve it, but since I'm inexperienced working in CIRCT, I would appreciate some advice on a) I've correctly identified the problem and b) possible solutions. 

@mikeurbach @mortbopet @andrewb1999 I would appreciate thoughts. Thanks!

mikeurbach, 2023-06-29:
Thanks for the simple reproducer and detailed spelunking! It seems like you are on the right track, but I haven't debugged this or done anything to confirm your hunch. If you're interested in tacking a crack at fixing it, perhaps a first thing to try would be swapping the order that patterns are added, so BuildWhileGroup is added to the list of patterns after BuildOpGroups. I have a feeling this probably falls over for other reasons, but might be worth a shot. Regardless, it seems more care needs to be taken in managing the blockSchedulables, since the order here is crucial.

mortbopet, 2023-06-30:
Thank you for reporting this and the detailed description. +1 for what Mike said - to make this even more robust, it might also be good if we implement a strict notion of "original ops in the block and their order <-> block scheduleables" so we aren't relying on pushing back on some vector in _hopefully_ the right order.

calebmkim, 2023-06-30:
Opened a PR to fix this bug #5526. 

Reordering the `BuildWhileGroup` and `BuildOpGroup` patterns didn't quite work. 
Instead, I made some small changes to allow me to add all the WhileScheduleable to the SchedulerInterface during the `BuildOpGroup` instead of the `BuildWhileGroup` pattern. This way, all the block scheduleables are added during the same pattern (`BuildOpGroup`) which fixes the ordering problem that we previously had.

PR #5526 (2023-07-06): [SCF-to-Calyx] Fix Ordering Bug 
Fixes #5521. 

The main thing was adding the `WhileScheduleable` to the `BlockScheduler` during the `BuildGroupOps` pattern instead of the `BuildWhileGroup` pattern, which means that all the block schedulables are added during the same pattern, preventing the order mismatching that was going on before. To do this, I needed to add `initGroups` (i.e., the groups that execute before the while loop that initialize the loop variables) to the `LoopLoweringStateInterface`. 

I also added a test case in `convert_controlflow.mlir`.