llvm/circt #5530: [FIRRTL] Verify that lowerToBind is never nested

seldridge, 2023-06-30:
If a circuit contains a bind-under-bind, then this can't be lowered to SystemVerilog without transformation. It would be simpler (for now) to make any nested bind a verifier error.

E.g., the following should be a verifier error:

```mlir
firrtl.circuit "Foo" {
  firrtl.module @Baz() {}
  firrtl.module @Bar() {
    firrtl.instance baz {lowerToBind} @Baz()
  }
  firrtl.module @Foo() {
    firrtl.instance bar {lowerToBind} @Bar()
  }
}
```

uenoku, 2023-07-01:
I was under the impression that nested bindings are illegal transitively (otherwise we can legalize nested bindings by creating wrappers), e.g. the following IR is also illegal. Is that true? If so, the verification would not be simple since we need verify in an inter-module manner. 

```mlir
firrtl.circuit "Foo" {
  firrtl.module @Baz() {}
  firrtl.module @Bar() {
    firrtl.instance baz {lowerToBind} @Baz()
  }
  firrtl.module @Bar_wrapper() {
    firrtl.instance bar @Bar()
  }
  firrtl.module @Foo() {
    firrtl.instance bar {lowerToBind} @Bar_wrapper()
  }
}
```

youngar, 2024-02-02:
Regarding @uenoku's idea, it is not accepted on Cadence (tested on EDA playground)
```systemverilog
// Top
module Top();
endmodule

module Top_LayerA();
  Component component();
endmodule

bind Top Top_LayerA top_layerA();

// Component
module Component();
endmodule

module Component_LayerA();
  wire w;
endmodule

bind Component Component_LayerA component_layerA();
```
```
TOOL:	xrun	20.09-s003: Started on Feb 02, 2024 at 18:24:46 EST
xrun: 20.09-s003: (c) Copyright 1995-2020 Cadence Design Systems, Inc.
	Top level design units:
		$unit_0x67f934e9
		Top
xmelab: *F,SVBTIS: Target module (Component instantiated at Top.top_layerA@Top_LayerA<module>.component) in bind statement (file: ./design.sv, line: 29) is source of another bind statement (file: ./design.sv, line: 19) which created instance (Top.top_layerA).
xrun: *E,ELBERR: Error during elaboration (status 2), exiting.
TOOL:	xrun	20.09-s003: Exiting on Feb 02, 2024 at 18:24:46 EST  (total: 00:00:00)
```