llvm/circt #5535: Nightlies failing

teqdruid, 2023-07-03:
https://github.com/llvm/circt/actions/runs/5444210245

Looks like we're missing a cmake dep.

And there's a failure in one of the ESI integration tests.

teqdruid, 2023-07-03:
On the ESI failure:

```
--
Command Output (stderr):
--
[ 11%] Creating directories for 'refl-cpp-populate'
[ 22%] Performing download step (git clone) for 'refl-cpp-populate'
Cloning into 'refl-cpp-src'...
remote: Internal Server Error
fatal: unable to access 'https://github.com/veselink1/refl-cpp/': The requested URL returned error: 500
Cloning into 'refl-cpp-src'...
remote: Internal Server Error
fatal: unable to access 'https://github.com/veselink1/refl-cpp/': The requested URL returned error: 500
Cloning into 'refl-cpp-src'...
remote: Internal Server Error
fatal: unable to access 'https://github.com/veselink1/refl-cpp/': The requested URL returned error: 500
-- Had to git clone more than once: 3 times.
CMake Error at refl-cpp-subbuild/refl-cpp-populate-prefix/tmp/refl-cpp-populate-gitclone.cmake:39 (message):
  Failed to clone repository: 'https://github.com/veselink1/refl-cpp'

make[2]: *** [CMakeFiles/refl-cpp-populate.dir/build.make:102: refl-cpp-populate-prefix/src/refl-cpp-populate-stamp/refl-cpp-populate-download] Error 1
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/refl-cpp-populate.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

CMake Error at /usr/share/cmake-3.23/Modules/FetchContent.cmake:1088 (message):
  Build step for refl-cpp failed: 2
Call Stack (most recent call first):
  /usr/share/cmake-3.23/Modules/FetchContent.cmake:1[217](https://github.com/llvm/circt/actions/runs/5444210245/jobs/9901739734#step:9:218):EVAL:2 (__FetchContent_directPopulate)
  /usr/share/cmake-3.23/Modules/FetchContent.cmake:1217 (cmake_language)
  /usr/share/cmake-3.23/Modules/FetchContent.cmake:1260 (FetchContent_Populate)
  CMakeLists.txt:16 (FetchContent_MakeAvailable)
```

This looks like a networking hiccup.

@mortbopet

uenoku, 2023-07-03:
Arcilator build error should be fixed by https://github.com/llvm/circt/commit/4ffcdd22c61a735a09766e065b39ac9f6d2125d8

teqdruid, 2023-07-03:
Sorry @uenoku . Didn't see that you fixed it and re-ran. I just file an issue every time i get a failure email two or so days in a row. Thanks for noticing and fixing!

closed 2023-07-03 by teqdruid

uenoku, 2023-07-03:
No worry! Thank you for always keeping nightly CI clean!