llvm/circt #5541: How to remove _GEN wires?

flexiot, 2023-07-06:
Here is a mux example. firtool generates some **_GEN_** wires as there are multiple callings.
[image]

But I would like to inline those GEN wires into the mux line just like the following code.

[image]

I tried to enable `--disable-opt` option. This option indeed can put GEN wires into the mux line, but this `--disable-opt` also can disable other optimization which is not expected.

[image]

Is it possible to just disable this small(not liked) optimization and keep other optimization?

uenoku, 2023-07-06:
> Is it possible to just disable this small(not liked) optimization and keep other optimization?

Currently there is no such an option. Multiply used expressions are spilled into temporary wires unless expressions are very simple (e.g. not, extract). I think it might make sense to create a lowering option to inline multiply used expressions but not sure how much practically useful it is. I guess the output size will easily grow exponentially. Just curios, why do you want to inline `GEN` for this case? Is there any tool that produces in worse result?

flexiot, 2023-07-06:
> > Is it possible to just disable this small(not liked) optimization and keep other optimization?
> 
> Currently there is no such an option. Multiply used expressions are spilled into temporary wires unless expressions are very simple (e.g. not, extract). I think it might make sense to create a lowering option to inline multiply used expressions but not sure the cost model and how much practically useful it is. I guess output sizes will easily grow exponentially. Just curios, why do you want to inline `GEN` for this case? Is there any tool that produces worse results?

Thanks for your reply. The reason to inline `GEN `is to make the generated system verilog more readable and more consistent with the chisel code as those `GEN `variables do not exist in the chisel code.

flexiot, 2023-07-06:
This is the related chisel test code.
[image]