llvm/circt #5543: [Calyx] Missing codeowner

mortbopet, 2023-07-06:
As it stands, there is no [codeowner](https://github.com/llvm/circt/blob/main/CODEOWNERS) for calyx and related passes. I've sensed that there's often a hot-potato problem when it comes to issues around the Calyx dialect (who's responsible for bug fixes, reviews, "reference the native compiler"-style answers and issues, ...). I think a lot of this stems from the fact that the CIRCT implementation of Calyx is still to some degree not a first-class-citizen in Calyx world (correct me if I'm wrong).

I think it would be beneficial for the CIRCT Calyx project that we define a clear set of codeowners. Alongside that, more elaborate documentation about how to use Calyx in CIRCT, a Calyx rationale, etc. etc. such that newcomers to the project have someone to go to, and somewhere to reference, when experimenting and trying things out.

CC @rachitnigam @mikeurbach @andrewb1999 @matth2k

rachitnigam, 2023-07-06:
Thanks for creating the issue @mortbopet! I think you're right that defining a set of codeowners would be useful and seems like a relative quick change to things. However, there are a couple of other things you mentioned that are worth discussing:

**Lowering Passes**. There is a question about ownership of the various lowering passes from and to Calyx which (I think?) were developed by you and @mikeurbach. @andrewb1999 is in the process of reimplementing them with the new loop scheduling and pipeline dialects which should substantially improve their reliability and coverage (in the sense of supported constructs in `affine` and `scf`). @andrewb1999 and @matth2k would probably be able to take over as codeowners for that "into Calyx" flow in the next couple of months? OTOH, the story for ownership of Calyx to FSM/HW is less clear to us since we don't really interact with that flow a whole lot.

**Core Dialect**. @cgyurgyik originally developed the dialect so we should keep him as a codeowner (if he'd like) and again, @andrewb1999, @matth2k, and I can be owners.

**Native Compiler**. We do have [documentation](https://docs.calyxir.org/fud/circt.html) on how to use the CIRCT dialect with the native compiler but you're right: adding this information to the rationale doc would be good.

**Long-term**. With @andrewb1999 and @matth2k's new HLS flow using CIRCT (which is also a long-term research project), I think the Calyx dialect will continue to be maintained for a while and hopefully, as more people contribute back to it, we can expand the set of owners.

mikeurbach, 2023-07-06:
I'm a fan of Calyx and I have had my hands on parts of the CIRCT Calyx implementation, so I'm happy to be a codeowner for the historical perspective. I know I've been slacking on some recent reviews, but in general I'm happy to at least review PRs. I'm still very much interested in HLS flows that use CIRCT and Calyx and static timing.

cgyurgyik, 2023-07-18:
I am also happy to continue reviewing.

mortbopet, 2023-07-19:
Awesome, glad to hear that there's volunteers 🤚🤚!

Codeowner info should be specified in https://github.com/llvm/circt/blob/main/CODEOWNERS

closed 2023-08-01 by rachitnigam