llvm/circt #5562: [FIRRTL][LowerToHW] An uninstantiated module can crash LowerToHW

seldridge, 2023-07-10:
I came across the following failure. The problem occurs if a module which is not under the main module in the instance graph includes reference type ports:

```mlir
firrtl.circuit "Foo" {
  firrtl.module @Foo() {}
  firrtl.module @Bar(in %a: !firrtl.uint<1>, out %b: !firrtl.probe<uint<1>>) {
    %0 = firrtl.ref.send %a : !firrtl.uint<1>
    firrtl.ref.define %b, %0 : !firrtl.probe<uint<1>>
  }
}
```

The failure trips an assert in `LowerToHW` [here](https://github.com/llvm/circt/blob/main/lib/Conversion/FIRRTLToHW/LowerToHW.cpp#L2337):

```
Assertion failed: ((!result || !isa<FIRRTLType>(result.getType())) && "Lowering didn't turn a FIRRTL value into a non-FIRRTL value"), function setLowering, file LowerToHW.cpp, line 2338.
```

dtzSiFive, 2023-07-11:
Well that's fun :).  We don't really support multiple public modules presently (and our use of it for "DUT" isn't quite right), but of course we should handle this neatly regardless, ideally with a more useful diagnostic as to what's unexpected/wrong.  Is the issue regarding firtool producing an error/crashing when fed this input, or is this indeed specifically about LowertoHW not doing something else with this (not sure what that would be)?

From pipeline viewpoint, LowerXMR should remove ref-type uses/ports (mostly), so I'd start there.  Offhand I'd guess the way it walks the instance graph (bottom-up) doesn't include unreachable modules so it doesn't find them.  I'm not sure what is intended with this example re:ref uses though, do we expect more ABI macros for `Bar` as well, should we resolve XMR's within each public module maybe, or is the use of public visibility not just incidental re:example?  (FWIW, the ABI avoids naming conflicts there only because we have a single public module to export information for)
If there's some optional instantiation from Foo to Bar then that of course is more fun :).

Anyway depending on what's intended/going on here this should be straightforward to fix, if there's multiple XMR hierarchies we want to resolve XMR's under that would be good to know and slightly changes how we want to approach addressing this.

Anyway, in short: what's the expected behavior here? Better error, or..? :+1:

seldridge, 2023-07-11:
Great analysis!

I'll distill your analysis to there being two issues here:

1. `LowerToHW` should assert (or straight up error) if it ever sees a probe type.  These are all supposed to be removed before this conversion.  (Arguably, `LowerXMR` is a mandatory pre-pass of `LowerToHW` that we only enforce via the firtool pipeline.)
2. `LowerXMR` should lower this. My reasoning here is that the above circuit is lowered all the way to Verilog (producing a module `Bar` in its output) if those ref ports are real ports. Whether or not ABI macros are produced is (in CIRCT) a function of whether or not the module is public. I think we can continue to just use that logic here, too. The spec should be tightened up here one way or another.

dtzSiFive, 2023-07-11:
> 1. `LowerToHW` should assert (or straight up error) if it ever sees a probe type.  These are all supposed to be removed before this conversion.  (Arguably, `LowerXMR` is a mandatory pre-pass of `LowerToHW` that we only enforce via the firtool pipeline.)

Yep, with the caveat that the force/release operations are handled by `LowerToHW`, the dest operand being replaced with an `sv.xmr.ref` + `unrealized_conversion_cast`.

> 2\. `LowerXMR` should lower this. My reasoning here is that the above circuit is lowered all the way to Verilog (producing a module `Bar` in its output) if those ref ports are real ports. Whether or [...]

Okay.  I don't know I follow your reasoning, but it does seem likely we can do something reasonable-ish here instead of crashing (modulo fixing up ABI/spec so this is safe and such-- not sure if there are other avenues for names to cause collisions in ABI macros but might be good to think that through while visiting, perhaps using a reserved character as separator or mangling?).
There are other places multiple top nodes will run into problems and not be handled well (or how one might expect, which may vary), especially re:interactions with annotations, as this is basically "multiple designs same file" territory.  Anyway I'll let you steer that/your ship, just trying to point out possible rocks lurking under the surface along that path.

Anyway, teaching LowerXMR to walk all modules is conceptually a small change and offhand I think should "work".  Just need to post-order the entire instance DAG instead of just from the main module-- although looks like our (FIRRTL)InstanceGraph does not make this as trivial as it sounds (may be overlooking something).  Anyway, sounds good, something for the TODO :+1: .

seldridge, 2023-07-11:
Another option is to make this circuit construction illegal. This is a bit annoying to verify as it requires an instance graph analysis. (It also requires passes to delete dead modules as opposed to letting this be orthogonally handled by another pass/garbage collection step.)

That said, FIRRTL circuits (circuits with a single top) have long been pushing towards FIRRTL packages (circuits with multiple tops). I agree completely with your assessment that this is uncharted territory for the most part and many things may break.