llvm/circt #5569: circt cloning Issue

ankrtlpixxel, 2023-07-12:
`$ git clone git@github.com:llvm/circt.git`

_Cloning into 'circt'..._
_git@github.com: Permission denied (publickey)._
_fatal: Could not read from remote repository._

_Please make sure you have the correct access rights_
_and the repository exists._

How to get past this issue ?

ankrtlpixxel, 2023-07-13:
@uenoku Thanks for the help, it worked.

But going ahead with the installation -> While running the ninja command after cloning CIRCT, I am getting the following error.
```
[2509/4084] Linking CXX shared library lib/libLTO.so.17git
ninja: build stopped: subcommand failed.
```
Can you suggest what needs to be done here ?

uenoku, 2023-07-13:
Hmm, not sure. Could you paste CMake command?

ankrtlpixxel, 2023-07-13:
Here it is :
```
cmake -G Ninja ../llvm \
    -DLLVM_ENABLE_PROJECTS="mlir" \
    -DLLVM_TARGETS_TO_BUILD="host" \
    -DLLVM_ENABLE_ASSERTIONS=ON \
    -DCMAKE_BUILD_TYPE=DEBUG \
    -DCMAKE_EXPORT_COMPILE_COMMANDS=ON
```

_Also while running CMake it shows it wasn't able to find some files while Failed status for some runs._

[image]
[image]

uenoku, 2023-07-13:
Thanks, so you are encountering errors in llvm build, right? I’d first try building llvm with Release mode instead of debug since the error seems to be while in the linking. If that doesn’t solve, I’d ask a question in llvm discourse.

ankrtlpixxel, 2023-07-14:
Thanks for the help @uenoku . I was able to successfully install the same.

closed 2023-07-14 by uenoku