llvm/circt #5634: [CI] Include file missing for macos python wheel build

rsetaluri, 2023-07-19:
This looks like a fresh issue?

https://github.com/llvm/circt/actions/runs/5598847526/jobs/10239048497#step:6:2011

mikeurbach, 2023-07-19:
Seems like a missing cmake `DEPENDS`, and not really specific to macos Python wheel build. I'll take a quick look and update here.

mikeurbach, 2023-07-19:
Looks like the upstream Arith dialect has been using interfaces for some time. I'm not sure why we just hit this, perhaps we've been lucky. But I believe any CMake library in CIRCT that includes sources that import Arith.h will need to declare a dependency on MLIRArithOpsInterfacesIncGen, to ensure the Arith interfaces would be generated before the sources are compiled.

rsetaluri, 2023-07-19:
Thanks @mikeurbach. Took a first pass at this on #5636. Seems like we probably can't deterministically trigger the faulty compilation, so not sure if we can test this directly.