llvm/circt #5640: [SV] Introduce SystemVerilog `string` type

mortbopet, 2023-07-20:
#5616 introduces a new, high(er) level way of defining format string and printing in CIRCT.
Currently, this has a somewhat simple lowering that _requires_ matching `PrintOp(FormatVerilogStringOp(...))` into a `sv.fwrite` operation.
The more correct way of doing this would be to have a separate lowering for each op, such that the `FormatVerilogStringOp` lowers to e.g. a `$sformat(var, ...)` which is then printed by e.g. `$fwrite`:
```sv
string var;
$sformat(var, ...)
$fwrite(fd, "%s", var)
```
**however** - this would require that said `$sformat` has a `var` to write to, which must have a sv `string` type. Currently, we have no way of describing nor emitting such an `sv::InOutStringType` which we'd presumably define by a `sv::StringOp`. 

If the above were defined, we would probably also like to modify a bunch of operations in the SV dialect which currently takes a `StringAttr` but in theory would accept a `sv::InOutStringType`.

teqdruid, 2023-07-20:
According to Bing's Chat, `string format = $sformatf("%d %d %d", 1, 2, 3);`.

mortbopet, 2023-07-21:
Still, that doesn't seem like it can make us avoid having the `string` type, seeing as that `format` name still needs to be passed to `$fwrite` somehow.

teqdruid, 2023-07-25:
Yeah, but for that we should just re-use hw.string. We use all of the other data types from hw in SV output and `$sformatf` doesn't require the string being passed in, so the model matches pretty perfectly. (Unless ChatGPT was wrong, though I'm pretty sure I checked.)