llvm/circt #5661: [Documentation] Lack of Labels for CIRCT Diagram on Home Page

ManelMCCS, 2023-07-23:
The current implementation of the home page in the project repository displays a CIRCT diagram without any labels. The same diagram accompanied by labels can also be found in the "Getting Started" documentation. This inconsistency poses challenges for new users and hampers the understanding of the CIRCT diagram's components and functionality.

seldridge, 2023-07-23:
This is confusing!

The fix should be to add the legend just before this line: https://github.com/llvm/circt-www/blob/3b30183a2859fda58e4181c9b4e26d5f2bf290d9/website/content/_index.md?plain=1#L33

ManelMCCS, 2023-07-24:
I will fix it and make a PR

ManelMCCS, 2023-07-25:
@mortbopet do you know the path for the image with the "correct" diagram?

mortbopet, 2023-07-27:
https://github.com/llvm/circt/tree/main/docs gets moved to the root directory of the site. So it's most likely 
> `<p align="center">[image]</p>`