llvm/circt #5664: [MSFT] Non-deterministic IR creation in DiscoverAppIDs

uenoku, 2023-07-24:
Found with `-DLLVM_ENABLE_REVERSE_ITERATION=On`.

```
FAIL: CIRCT :: Dialect/MSFT/module_instance.mlir (33 of 601)
******************** TEST 'CIRCT :: Dialect/MSFT/module_instance.mlir' FAILED ********************
Script:
--
: 'RUN: at line 1';   /scratch/hidetou/circt/build_debug/bin/circt-opt /scratch/hidetou/circt/test/Dialect/MSFT/module_instance.mlir -verify-diagnostics | /scratch/hidetou/circt/build_debug/bin/circt-opt -verify-diagnostics | /scratch/hidetou/circt/build_debug/bin/FileCheck /scratch/hidetou/circt/test/Dialect/MSFT/module_instance.mlir
: 'RUN: at line 2';   /scratch/hidetou/circt/build_debug/bin/circt-opt /scratch/hidetou/circt/test/Dialect/MSFT/module_instance.mlir --msft-discover-appids -verify-diagnostics | /scratch/hidetou/circt/build_debug/bin/circt-opt -verify-diagnostics | /scratch/hidetou/circt/build_debug/bin/FileCheck /scratch/hidetou/circt/test/Dialect/MSFT/module_instance.mlir --check-prefix=APPID
: 'RUN: at line 3';   /scratch/hidetou/circt/build_debug/bin/circt-opt /scratch/hidetou/circt/test/Dialect/MSFT/module_instance.mlir --lower-msft-to-hw -verify-diagnostics | /scratch/hidetou/circt/build_debug/bin/circt-opt -verify-diagnostics | /scratch/hidetou/circt/build_debug/bin/FileCheck /scratch/hidetou/circt/test/Dialect/MSFT/module_instance.mlir --check-prefix=HWLOW
--
Exit Code: 1

Command Output (stderr):
--
/scratch/hidetou/circt/test/Dialect/MSFT/module_instance.mlir:33:17: error: APPID-LABEL: expected string not found in input
// APPID-LABEL: msft.module @Foo {WIDTH = 1 : i64} () -> (x: i32) attributes {childAppIDBases = ["extern", "foo"]}
                ^
<stdin>:2:73: note: scanning from here
 msft.module @top {} () attributes {childAppIDBases = ["extern", "foo"]} {
                                                                        ^
<stdin>:13:2: note: possible intended match here
 msft.module @Foo {WIDTH = 1 : i64} () -> (x: i32) attributes {childAppIDBases = ["foo", "extern"]} {
 ^

Input file: <stdin>
Check file: /scratch/hidetou/circt/test/Dialect/MSFT/module_instance.mlir

-dump-input=help explains the following input dump.

Input was:
<<<<<<
            1: module { 
            2:  msft.module @top {} () attributes {childAppIDBases = ["extern", "foo"]} { 
label:33'0                                                                             X~~ error: no match found
            3:  %foo.x = msft.instance @foo @Foo() {msft.appid = #msft.appid<"foo"[0]>} : () -> i32 
label:33'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            4:  %true = hw.constant true 
label:33'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~
            5:  %extern.out = msft.instance @extern @Extern(%true) <param: i1 = false> : (i1) -> i1 
label:33'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            6:  msft.output 
label:33'0     ~~~~~~~~~~~~~
            7:  } 
label:33'0     ~~~
            8:  msft.module @B {WIDTH = 1 : i64} (%a: i4) -> (nameOfPortInSV: i4) attributes {fileName = "b.sv"} { 
label:33'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            9:  %0 = comb.add %a, %a : i4 
label:33'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~
           10:  msft.output %0 : i4 
label:33'0     ~~~~~~~~~~~~~~~~~~~~~
           11:  } 
label:33'0     ~~~
           12:  msft.module @UnGenerated {DEPTH = 3 : i64} (%a: i1) -> (nameOfPortInSV: i1) 
label:33'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
           13:  msft.module @Foo {WIDTH = 1 : i64} () -> (x: i32) attributes {childAppIDBases = ["foo", "extern"]} { 
label:33'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
label:33'1      ?                                                                                                     possible intended match
           14:  %true = hw.constant true 
label:33'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~
           15:  %extern.out = msft.instance @extern @Extern(%true) <param: i1 = false> {msft.appid = #msft.appid<"extern"[0]>} : (i1) -> i1 
label:33'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
           16:  %c1_i4 = hw.constant 1 : i4 
label:33'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
           17:  %b.nameOfPortInSV = msft.instance @b @B(%c1_i4) {msft.appid = #msft.appid<"foo"[0]>} : (i4) -> i4 
label:33'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
           18:  %c0_i32 = hw.constant 0 : i32 
label:33'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            .
            .
            .
>>>>>>

--

********************
********************
Failed Tests (1):
  CIRCT :: Dialect/MSFT/module_instance.mlir

```
An array attribute is created based on the order of hash maps `appBaseCounts` and `localAppIDBases`.
https://github.com/llvm/circt/blob/f453a87cd79c0bcbbea7dc2c1df977321b7e9aa3/lib/Dialect/MSFT/Transforms/MSFTDiscoverAppIDs.cpp#L94-L105

teqdruid, 2023-07-25:
I'm pretty sure this is fixed now. I've tested it with and without the reverse iteration flag.

teqdruid, 2023-07-25:
Thanks for reporting this @uenoku !

commit 157008ca (2023-07-25): [MSFT] Fix non-deterministic order in appid discovery

Fixes #5664
closed 2023-07-25 by teqdruid